nanoframework/Home #527: USB CDC support

networkfusion, 2019-09-14:
USB is gaining traction, even in the embedded world. However, most sensors that utilise it still use serial over USB (i.e. USB CDC). UWP apps support this using the following function:

```
string deviceSelector = Windows.Devices.SerialCommunication.SerialDevice.GetDeviceSelectorFromUsbVidPid(device.vid,device.pid);
```

nanoFramework already supports serial communication, and ChibiOS has a community contribution for it (which I believe nf is using for USB storage on STM32 targets) www.chibios.com/forum/viewtopic.php?f=16&t=2968&start=40

josesimoes, 2019-09-16:
@networkfusion was not clear to me if you are asking for the ability to expose the nF device as a CDC device or if it's for the nF device to become USB host in order to communicate with USB CDC devices.

networkfusion, 2019-09-16:
The latter `nF device to become USB host in order to communicate with USB CDC devices`

networkfusion, 2019-09-30:
@josesimoes , given https://github.com/nanoframework/ChibiOS-Contrib/blob/master/os/hal/src/usbh/hal_usbh_ftdi.c what is the best way of implementing it in line with to https://docs.microsoft.com/en-us/uwp/api/windows.devices.serialcommunication.serialdevice.getdeviceselectorfromusbvidpid ?

josesimoes, 2019-10-01:
@networkfusion still haven't looked into this. 
For the MSD I used the code from ChibiOS community. I recall that CDC was there too.
We should check which one is offering the best implementation: the official repo or the community one.

networkfusion, 2019-10-01:
@josesimoes yes the contrib repo is the only place with usb cdc as mentioned above. I have added a PR in the nf-interpreter repo which does all of the skeleton.  I just need to work out the best way of linking it to serial communication...

josesimoes, 2019-12-02:
still active, right @networkfusion 😁  ?

closed 2020-02-07 by stale[bot]