nanoframework/Home #528: NucleoF411RE  after disconnect from power it do not start more and have to re flash nanoframework

bmecdev, 2019-09-25:
## Details about Problem

**Target NucleoF411RE:**

**Firmware image version: ST_NUCLEO64_F411RE_NF-1.1.187.0**

## Description

 I'm working with NucleoF411RE and have  issues after disconnect from power it do not start more and have to re flash nanoframework

## Detailed repro steps so we can see the same problem

1. flash only nanobooter to board and led start blinking.

2. look at visual studio on device explorer and the board is visible.

3. disconnect the board from usb getting power off an reconnect it.

4. look at visual studio on device explorer and the board is visible.

5. flash nanoCLR to the board and led stops blinking.

6. look at visual studio on device explorer and the board is visible.

7. disconnect the board from usb getting power off an reconnect it.

8. look at visual studio on device explorer and the board is not visible.

And I have to re flash nanoframework to watch the board again in visual studio device explorer

josesimoes, 2019-09-25:
Missing above: the VS version and extension version that you are using.

josesimoes, 2019-09-25:
Also, if you can get that from the 1st boot, paste here the device capabilities output.

And another question: are you deploying any managed app after the 1st boot? Or what you are describing occurs with just the booter+CLR flashed on the board?

bmecdev, 2019-09-25:
Hi I'm using Visual Studio enterprise 2017 version 15.9.14  and
nanoframework extension version 1.2.5.31

Yes it occurs with just the booter+CLR flashed on the board

Device capabilities

System Information
HAL build info: 1.1.187.0, nanoCLR running @ ST_NUCLEO64_F411RE_NF built
with ChibiOS v19.1.3.1
Target: ST_NUCLEO64_F411RE_NF
Platform: STM32F4

Image build @ Sep  8 2019 GNU ARM GCC v8.3.1

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Solution Build Info: 1.1.187.0, nanoCLR running @ ST_NUCLEO64_F411RE_NF
built with ChibiOS v19.1.3.1

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.2.2.0, checksum 0x6AC9143D
  nanoFramework.Runtime.Native v100.0.5.0, checksum 0x153CD2E3
  nanoFramework.Runtime.Events v100.0.6.0, checksum 0x9E949BB1
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  Windows.Devices.Adc v100.1.3.1, checksum 0xA84B23D2
  Windows.Devices.Gpio v100.1.1.0, checksum 0xD7F65BB7
  Windows.Devices.I2c v100.2.0.0, checksum 0x100B5D05
  Windows.Devices.Pwm v100.1.3.1, checksum 0x8E93FB40
  Windows.Devices.SerialCommunication v100.1.0.0, checksum 0x4B336BE9
  Windows.Devices.Spi v100.1.3.0, checksum 0xBDDC51FB

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x20000000  0x00020000
  FLASH 0x08000000  0x00080000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x08000000       1      0x004000     nanoBooter
      1    0x08004000       3      0x004000     nanoCLR
      2    0x08010000       1      0x010000     nanoCLR
      3    0x08020000       2      0x010000     nanoCLR
      4    0x08040000       4      0x010000     Deployment

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x08000000    0x004000 (16kB)     nanoBooter
  0x08004000    0x03C000 (240kB)    nanoCLR
  0x08040000    0x040000 (256kB)    Deployment

Deployment Map
Empty

-- 
---------------------------------------------------------------
Camilo Botero Sierra
Ingeniero Mecatrónico
Gerente general
Botero Mechatronics SAS
---------------------------------------------------------------

Contacto:

Cel: 3227006045
Cl. 187 #46-55, Bogotá
Email: cbotero@boteromechatronics.com
Visítanos en: http://www.boteromechatronics.com/
------------------------------------------------------------------------

jabacz, 2019-09-26:
I can confirm this. Probably it's the same as the [issue 526](https://github.com/nanoframework/Home/issues/526) 
I made more investigation. I can confirm exactly the issue of device dying after power off on Surface Pro6 (Intel I7). At desktop PC with Ryzen 1600X the device dies after any communication from Visual Studio (ping, device capabilities, deploy). It's a bit puzzling when with Intel you can make it work (blinky) till power off, while with AMD CPU it's enough to "ping" the device to kill it.
Visual studio tested 2017 as well as 2019 with NF 1.2.5.x and device FW #187
In both cases re-flashing makes is temporarily work.

jabacz, 2019-10-01:
Update: Upgrading VS2019 extension to 1.2.5.3 makes RYZEN desktop work the same was as described in this issue post. Device works till powered off. One part of the puzzle seems solved :-)

josesimoes, 2019-10-01:
@jabacz thanks for the update.
The investigation of the root cause is still on the queue, please be patient.

valoni, 2019-10-28:
SWO kill Serial Communication 
**- (need to compile with SWO_OUTPUT="OFF") -**

tagelv, 2019-11-10:
I confirm this behavoir and in addition I want to provide my trick to avoid re-flashing the target board:
1. Plug the target board to USB
2. Open ST-LINK Utility and connect to the target
3. Press User botton and Reset button at the same time (or first time - user button and almost immediately - reset button)

Result: flashed software runs, Device Explorer displays plugged target

jabacz, 2019-11-26:
@tagelv  This does not help with the issue, unfortunately. DEPLOY of DEBUG, or any Visual Studio invoked operation do not work.

piwi1263, 2019-11-26:
@tagelv @jabacz as @valoni mentioned above please have the SWO_OUTPUT set to OFF in your variants file, clear your build folder and recompile/build otherwise it WILL not work. This behavior is looked at and until that is resolved you will have to keep it that way.

closed 2019-12-26 by josesimoes