nanoframework/Home #529: Garbage Collection Heap Compaction bug

MateuszKlatecki, 2019-09-25:
When calling Debug.GC(true) (forcing garbage collection with heap compactation ) the execution stops with hard fault in some cases. I noticed this behavior on iMXRT1062 with custom application (about 11k lines of code). For now I cannot reproduce it with synthetic test app. 
If the GC is forced without the heap compactation it will perform GC without any issues.

As a temporary solution, we've disabled Heap Compaction globally by setting the NoCompaction flag during CLR startup.

josesimoes, 2019-09-25:
I've noticed the above when making that call right at the start of `Program.Main()`. If that same call is made further down when the execution has started it seems to work.

I don't have any explanation for the fact. The only thing that I could think of is that, for some reason, it requires the thread spawning to have started... But, then again, this could be just a coincidence....

josesimoes, 2019-11-14:
@MateuszKlatecki have you had the chance to test the above? If you can confirm that behaviour I think this can be closed as there is no (apparently) obvious cause for it.

josesimoes, 2020-05-07:
Hard fault ~ line 279 @ src\CLR\Core\GarbageCollector_ComputeReachabilityGraph.cpp

josesimoes, 2020-05-07:
@MateuszKlatecki this code executes on `ExecuteGarbageCollection()`.
Are you forcing the GC execution or is it called regularly on your application execution?

morali, 2020-05-07:
@josesimoes 
We are calling regularly GC every couple of seconds without forcing heap compaction.

This hardfault is very irregular. The "sub" pointer points to some odd location instead of managed heap.,

[image]

josesimoes, 2020-05-07:
May I ask why you're calling it?

morali, 2020-05-08:
Yes, AFAIK there's no method of getting free memory, besides calling GC.

josesimoes, 2020-05-08:
Got you.
So that's clearly a new feature. Please add that to the runtime API project as a suggestion.

josesimoes, 2020-07-09:
Closing this has it hasn't had any updates nor progresses.

closed 2020-07-09 by josesimoes