nanoframework/Home #530: static variables are not initialized

adaslesniak, 2019-10-16:
Please read the following information before posting the issue.

## Details about Problem

**nanoFramework area:**  compilator? clr?

**VS version:** professional 2017 v.15.9.16

**VS extension version>:** 1.2.5.31

## Description
static variable assigned in place where it is declared isn't initialized when accessing it.

. . .
private static ILogImplementation implementation = new FirmwareLog(); //part of static class
. . . 
public static void SetTreshold(string onChannel, LogLevel treshold) {
    implementation.SetChannel(onChannel, treshold); // throws null reference excpetion
}

## Detailed repro steps so we can see the same problem

it's little hard for me now to find time to cut this to core simple example with this behaviour, but this situation happened few times in various places when I used static variables and assumed they will be always initialized

## Expected behaviour
if static variable is assigned in place of declaration it should never be null

josesimoes, 2019-10-16:
This has been reported before. For example:

https://github.com/nanoframework/Home/issues/383#issuecomment-416199049 

@adaslesniak could you please try to move that class to a separate assembly to verify the above?

josesimoes, 2020-04-28:
Tested in STM32F769 DISCO, dev caps as follows and couldn't reproduce.

If you're still experiencing this, please reopen and add a test project showing this.

```
System Information
HAL build info: nanoCLR running @ ST_STM32F769I_DISCOVERY
  Target:   ST_STM32F769I_DISCOVERY
  Platform: STM32F7

Firmware build Info:
  Date:     Apr 28 2020
  Type:     MinSizeRel build with ChibiOS v19.1.3.1
  Version:  1.4.0.0
  Compiler: GNU ARM GCC v9.2.1

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.4.8.0, checksum 0xE988B5AD
  nanoFramework.Runtime.Native v100.0.6.1, checksum 0xBD754FED
  nanoFramework.Hardware.Stm32 v100.0.4.4, checksum 0xE171FDE1
  nanoFramework.Devices.Can v100.0.5.0, checksum 0xF319D40E
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xA5C172BD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.7.1, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.4, checksum 0x39DA4F21
  System.Net v100.1.3.1, checksum 0x1118F266
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  System.Devices.Dac v100.0.0.6, checksum 0x5ED97E79
  Windows.Devices.Gpio v100.1.2.1, checksum 0x9A7A2EA4
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  Windows.Devices.SerialCommunication v100.1.1.1, checksum 0x82260711
  Windows.Devices.Spi v100.1.4.1, checksum 0x59B5BFC3
  Windows.Storage v100.0.2.0, checksum 0x5160A7B6

```

closed 2020-04-28 by josesimoes