nanoframework/Home #531: Error deploying to ST_NUCLEO64_F091RC

Jamesso, 2019-10-20:
## Details

**VS version:** 2017 v15.9.16

**VS extension version:** 1.2.5.38

**Target:** STM32NUCLEO-F091RC

**Firmware image version:**
ST_NUCLEO64_F091RC-1.3.0
ST_NUCLEO64_F091RC-1.3.2
ST_NUCLEO64_F091RC-1.4.0-preview.5

**Worked before? If so, with which nanoFramework image version:**
ST_NUCLEO64_F091RC-1.0.3

## Description

Cannot deploy Blinky sample to ST_NUCLEO64_F091RC. Output shows an error writing to device memory.

## Steps

1. Install latest stable or preview versions.

2. Download and build Blinky sample.

3. Deploy to ST_NUCLEO64_F091RC.

## Expected Behavior

Deployment should be successful.

## Device Capabilities

```
System Information
HAL build info: 1.4.0.0, nanoCLR running @ ST_NUCLEO64_F091RC built with ChibiOS v19.1.3.1
Target: ST_NUCLEO64_F091RC
Platform: STM32F0

Image build @ Oct 17 2019 GNU ARM GCC v5.4.1

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Solution Build Info: 1.4.0.0, nanoCLR running @ ST_NUCLEO64_F091RC built with ChibiOS v19.1.3.1

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.2.3.0, checksum 0x91CE6848
  nanoFramework.Runtime.Native v100.0.5.0, checksum 0x153CD2E3
  nanoFramework.Hardware.Stm32 v100.0.4.3, checksum 0x3E87EA8A
  nanoFramework.Runtime.Events v100.0.6.0, checksum 0x9E949BB1
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  Windows.Devices.Gpio v100.1.1.0, checksum 0xD7F65BB7
  Windows.Devices.I2c v100.2.0.0, checksum 0x100B5D05
  Windows.Devices.Pwm v100.1.3.1, checksum 0x8E93FB40
  Windows.Devices.SerialCommunication v100.1.0.0, checksum 0x4B336BE9
  Windows.Devices.Spi v100.1.3.0, checksum 0xBDDC51FB

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x20000000  0x00008000
  FLASH 0x08000000  0x00040000

Empty

Deployment Map
Empty
```

## Error Output

```
2019-10-17 18:56:31Z [Check and start debug engine on nanoDevice.]
2019-10-17 18:56:31Z [Connect successful.]
2019-10-17 18:56:31Z [Erase deployment area successful.]
2019-10-17 18:56:32Z [Computing deployment blob.]
2019-10-17 18:56:33Z [Deploying assemblies.]
2019-10-17 18:56:33Z [Incrementally deploying assemblies to the device]
2019-10-17 18:56:39Z [Error writing to device memory @ 0x0802A000 (2048 bytes). Error code: 0.]
2019-10-17 18:56:39Z [Error deploying assemblies to the device.]
2019-10-17 18:56:40Z [Deploying assemblies. Second attempt.]
2019-10-17 18:56:40Z [Incrementally deploying assemblies to the device]
2019-10-17 18:56:46Z [Error writing to device memory @ 0x0802A000 (2048 bytes). Error code: 0.]
2019-10-17 18:56:46Z [Error deploying assemblies to the device.]
2019-10-17 18:56:46Z [Deployment failed.]
2019-10-17 18:56:46Z [Exception occurred during deployment.
 Deploy failed. 
  
    at nanoFramework.Tools.VisualStudio.Extension.DeployProvider.<>c__DisplayClass17_1.<<DeployAsync>b__4>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd(Task task)
   at nanoFramework.Tools.VisualStudio.Extension.DeployProvider.<DeployAsync>d__17.MoveNext()]
```

josesimoes, 2019-10-25:
Confirmed issue. Seems to be an bug in the flash driver.

Jamesso, 2019-10-31:
Now running 1.4.0-preview.17 and still have the same error.

josesimoes, 2019-11-13:
An hard fault is occurring when erasing the device flash thus causing the device to become irresponsive. Requires further investigation.

closed 2019-12-02 by josesimoes