nanoframework/Home #534: FileIO.ReadText returns empty string even when the file is not empty

hololite, 2019-10-21:
Environment: STM32F76xx with micro sd card, nanoclr 1.3.0.3.
(Note: this issue is seen on microsd file, I did not test it on flash)
Repro steps:
- On the micro sd card, create a simple file with a few text using notepad or vi on your computer.
- Place the sd card on the board
- Try to read the file using FileIO.ReadFile()
- The result is empty string

Another way to test it is to run the Scenario3_WriteAndReadTextInAFile.cs from sample repo.
The last section of the codes compares the written and read strings and they are different, which is not expected.

hololite, 2019-10-21:
Typo above. Should use the FileIO.ReadText API to read the file.

josesimoes, 2019-10-22:
This occurs in the STM32F7 series because of caching issues. It works just fine on STM32F4 and other that don't have cache.

Issue has been raised in ChibiOS forum and a possible fix discussed. Waiting from a follow from their end. Post [here](http://www.chibios.com/forum/viewtopic.php?f=16&t=5025).

josesimoes, 2019-11-11:
Update from ChibiOS issue and not what we would like to hear: they wont be providing a fix for this. 

This requires patching or a fix implemented at our end.

networkfusion, 2019-11-11:
@josesimoes I would recommend we either use the version in the contrib repo and patch that, or have a seperate repo for fatfs (perhaps shared with freertos targets... thoughts?

networkfusion, 2019-11-11:
It might also be worth getting in touch with the author of fatFS to see if he would like to fix the core officially

josesimoes, 2019-11-11:
The required fix is very target specific so I don't thing there is nothing to be "fixed" upstream…
It's really up to us.

My 1st thought is to have our FatFS glue layer for ChibiOS to take care of these tiny details.

networkfusion, 2020-01-20:
@josesimoes any further thoughts on this?

josesimoes, 2020-01-21:
At this times I think the best option is to patch the (already) modified Fats from ChibiOS. The remedy is clear: clear and flush cache on write and read operations. Exactly the same that what coded for the Spiffs driver for F7 (see Orgpal3 board).

The root cause is definitely the combination of DMA and F7 cache. DMA can't be disabled, so...

networkfusion, 2020-01-21:
PR submitted for this, but needs checking against targets other than F7...

networkfusion, 2020-02-06:
There is some good advice on how to align DMA buffers in the description of http://elm-chan.org/fsw/ff/doc/dread.html

networkfusion, 2020-04-06:
Still valid!

networkfusion, 2020-05-14:
Reading files has been no issues for me in quite a while! however I have not been able to test writes. Is this still an issue???

gligorov, 2020-05-14:
Yes, the latest image when using write, it fails in some cases.  For example, I had to add some pause to make it work with the latest image:

 var myFile = media.CreateFile($"BackUP_{DateTime.UtcNow.Millisecond}.txt", CreationCollisionOption.ReplaceExisting);
                        Thread.Sleep(256);
                        FileIO.WriteText(myFile, currentConfig);

without the extra timeout between these calls, the WriteText would fail.

josesimoes, 2020-07-20:
Still valid

networkfusion, 2021-09-10:
Needs revisiting with the new API and later targets.

networkfusion, 2021-10-13:
`Windows.Storage.Devices` will soon be deprecated. This issue needs re-evaluating for `System.IO.FileSystem`.

networkfusion, 2022-06-17:
[image]

closed 2024-07-23 by josesimoes