nanoframework/Home #535: Opening Device Explorer Crashes Windows

RobLawlor2, 2019-10-22:
Hello,
I have just attempted to use the nano Framework , however I get a system crash(blue screen) when I attempt to open device explorer.
I can open templates for the nano framework and build them.

I get this system crash with both VS2019 and VS2017.

Microsoft Visual Studio Professional 2019 (2)
Version 16.3.5
VisualStudio.16.Release/16.3.5+29411.108
Microsoft .NET Framework
Version 4.8.03761

nanoFrameWork  version 1.2.5.8

josesimoes, 2019-10-22:
What is the target device you are using?

RobLawlor2, 2019-10-22:
I have no target device as of yet. A STM32 Nucleo-144  (STM32F746ZGT6)  should be arriving tomorrow.
However the issue is not with attmpting to connect, rather the device explorer window does not open and windows crashes.

josesimoes, 2019-10-22:
🤔 the only explanation that occurs to me is that one of the COM ports connected is misbehaving... 
We had reports before on failed auto-detection with BLE virtual COM ports. But nothing like that.

The only BSoD report was related with outdated versions of FTDI drivers and ESP32 target boards.h
Any of the above helps?

RobLawlor2, 2019-10-22:
Hi ,
I had some virtual Com ports setup using com0com.
I uninstalled them and the device explorer opened without crashing.
Thank you very much. I'm looking forward to using the nano framework now.

josesimoes, 2019-10-22:
Nice!

Glad that you've managed to get past this.
Wish you a lot of fun with nanoFramework.

Make sure you join our friendly [Discord community](https://discord.gg/gCyBu8T). 😉

josesimoes, 2019-10-22:
If you could provide the VID/PID or other USB ID that can be used to identify those virtual COM ports that would allow us to add those to the exclusion list so they are skipped during the detection.

closed 2019-10-22 by josesimoes

RobLawlor2, 2019-10-22:
Hi, the IDs are as follows:

root\com0com
com0com\port
com0com\comport

thanks again.