nanoframework/Home #536: CreateFile Throws Exception if Existing File Needs to be Overwritten on SPI Flash

gligorov, 2019-10-22:
## Details about Problem
Two scenarios that throw exception, not sure if they are one but here are both:  This is on STM32 build, F7 usign internal SPI Flash as storage.  

Case 1:
Running the Samples - Storage, on Internal Storage, SPI Flash in my case, the system passes all tests but Case 9, where it throws this exception:

== Scenario9_RenameFile ==
    #### Exception System.Exception - CLR_E_FILE_NOT_FOUND (1) ####
    #### Message: 
    #### Windows.Storage.StorageFile::GetFileFromPathNative [IP: 0000] ####
    #### Windows.Storage.StorageFile::GetFileFromPath [IP: 0015] ####
    #### FileAccess.Scenario9_RenameFile::Execute [IP: 0017] ####
    #### FileAccess.Program::Main [IP: 00cd] ####
Exception thrown: 'System.Exception' in Windows.Storage.dll
    #### Exception System.Exception - CLR_E_FILE_NOT_FOUND (1) ####
    #### Message: 
    #### Windows.Storage.StorageFile::GetFileFromPathNative [IP: 0000] ####
    #### Windows.Storage.StorageFile::GetFileFromPath [IP: 0015] ####
    #### FileAccess.Scenario9_RenameFile::Execute [IP: 0031] ####
    #### FileAccess.Program::Main [IP: 00cd] ####
Exception thrown: 'System.Exception' in Windows.Storage.dll
ERROR: Can't rename file :Exception was thrown: System.Exception

Case 2:
My own, code, I write a file to internal storage  - first time file does  not exist so all is well.  Next time if I want to write that same file - overwrite existing one, it fails with the exception below.

code used that throws exception

  var file = systemFlash.CreateFile(confFileName.ToLower(), CreationCollisionOption.ReplaceExisting);

It throws System Exception, StackTrace:  Could not evaluation expression

josesimoes, 2019-10-25:
Check updated Storage sample, in particular Scenario 9.

https://github.com/nanoframework/Samples/commit/c1500f684a9d342c06c5940d4bda9223e0185968

gligorov, 2019-12-05:
Latest build in the repo, after fix for the SD Card/USB Flash use at the same time, the exception is back for Internal storage.  To test, simply run the Storage samples, twice in a row, and the Exception is thrown.   https://media.discordapp.net/attachments/484073476626317333/652157989847040001/unknown.png

josesimoes, 2019-12-06:
@gligorov can't reproduce this... 
I've loaded the latest image in ORGPAL_PALTHREE and I can run the storage sample over and over without ANY exception...

gligorov, 2019-12-06:
@josesimoes , did again a new build, got the latest Storage samples and problem is still there.  If I use my prior image, all goes well, see below.   If I use newest image, same board same samples, exception is thrown, see pic.

Prior Image Runs fine:
== GetRemovableDevices ==
ERROR: Can't do anything. There is no removable device present.
== GetInternalDevices ==
== Scenario2_CreateAFileInStorage ==
OK: Successfully created file: I:\file1.txt
== Scenario3_WriteAndReadTextInAFile ==
OK: Successfully created file: I:\file2.txt
OK: read text matches written text.
== Scenario4_WriteAndReadBytesInAFile ==
OK: Successfully created file: I:\file3.bin
Of the 0 in buffer, 0 remains to be written to 'file3.bin':
this is a string to be saved as binary data
The following 256 bytes of text were read from 'file3.bin':
this is a string to be saved as binary data
== Scenario9_RenameFile ==
The program '[0x1] nanoFramework application: Managed' has exited with code 0 (0x0).

Newest Image Throws Exception
== GetRemovableDevices ==
ERROR: Can't do anything. There is no removable device present.
== GetInternalDevices ==
== Scenario2_CreateAFileInStorage ==
    ++++ Exception System.Exception - CLR_E_FILE_IO (1) ++++
    ++++ Message: 
    ++++ Windows.Storage.StorageFolder::CreateFileNative [IP: 0000] ++++
    ++++ Windows.Storage.StorageFolder::CreateFile [IP: 0006] ++++
    ++++ FileAccess.Scenario2_CreateAFileInStorage::Execute [IP: 0011] ++++
    ++++ FileAccess.Program::Main [IP: 00bb] ++++
Exception thrown: 'System.Exception' in Windows.Storage.dll
Error: Unable to create file: Exception was thrown: System.Exception
== Scenario3_WriteAndReadTextInAFile ==
    ++++ Exception System.Exception - CLR_E_FILE_IO (1) ++++
    ++++ Message: 
    ++++ Windows.Storage.StorageFolder::CreateFileNative [IP: 0000] ++++
    ++++ Windows.Storage.StorageFolder::CreateFile [IP: 0006] ++++
    ++++ FileAccess.Scenario3_WriteAndReadTextInAFile::Execute [IP: 0012] ++++
    ++++ FileAccess.Program::Main [IP: 00c1] ++++
Exception thrown: 'System.Exception' in Windows.Storage.dll
An unhandled exception of type 'System.Exception' occurred in Windows.Storage.dll
[image]

gligorov, 2019-12-06:
Forgot to mention that I am building the latest image using GCC tools 9. Prior image is built GCC 8.

josesimoes, 2019-12-06:
Could you please try with the image built from Azure Pipeline?

Using `nanoff --update --target ORGPAL_PALTHREE`

gligorov, 2019-12-06:
Yes, I just tried and is working perfectly, so maybe something about my local build or build options? Thanks for clearing it up, I just need to figure out what am I doing wrong.

closed 2019-12-06 by gligorov

networkfusion, 2019-12-06:
@gligorov try cleaning your build folder...

gligorov, 2019-12-06:
Well the mistery continues. If I build my local build with all options on, it works. If I do a build without Network/SSL then I get the problem.  So, thanks @networkfusion and @josesimoes , but not sure about it, and I will do few more tests.