nanoframework/Home #537: Wifi Sample fails to load on ESP32

lafe, 2019-10-27:
## Details about Problem

**nanoFramework area:** Samples

**VS version:** 2019

**VS extension version:** 1.2.5.8

**Target:** ESP32

**Firmware image version:** 1.3.2.3 (installed using `nanoff --stable --target ESP32_WROOM_32 --update --serialport COM3`)

**Device capabilities output:**
```
System Information
HAL build info: 1.3.2.3, nanoFramework running ESP32 built with IDF v3.1
Target: ESP32_WROOM_32
Platform: ESP32

Image build @ Oct 17 2019 GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Solution Build Info: 1.3.2.3, nanoFramework running ESP32 built with IDF v3.1

AppDomains:

Assemblies:
  ScanWiFi, 1.0.0.0
  Windows.Devices.Wifi, 1.0.7.3
  mscorlib, 1.3.0.3
  nanoFramework.Runtime.Events, 1.3.0.3

Native Assemblies:
  mscorlib v100.2.3.0, checksum 0x91CE6848
  nanoFramework.Runtime.Native v100.0.5.0, checksum 0x153CD2E3
  nanoFramework.Hardware.Esp32 v100.0.7.0, checksum 0xC16385A2
  nanoFramework.Devices.OneWire v100.0.3.3, checksum 0xC7BEAEBC
  nanoFramework.Networking.Sntp v100.0.4.3, checksum 0xB617FDFB
  nanoFramework.Runtime.Events v100.0.6.0, checksum 0x9E949BB1
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.3, checksum 0xC5A2A195
  System.Net v100.1.3.0, checksum 0x0B718489
  Windows.Devices.Adc v100.1.3.1, checksum 0xA84B23D2
  Windows.Devices.Gpio v100.1.1.0, checksum 0xD7F65BB7
  Windows.Devices.I2c v100.2.0.0, checksum 0x100B5D05
  Windows.Devices.Pwm v100.1.3.1, checksum 0x8E93FB40
  Windows.Devices.SerialCommunication v100.1.0.0, checksum 0x4B336BE9
  Windows.Devices.Spi v100.1.3.0, checksum 0xBDDC51FB
  Windows.Devices.Wifi v100.0.6.0, checksum 0x5AC050C9
  Windows.Storage v100.0.1.0, checksum 0x8D8C8026

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x200000c0  0x00020000
  FLASH 0x08000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x110000     nanoCLR
      1    0x00120000       1      0x1B0000     Deployment
      2    0x002D0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x002D0000    0x040000 (256kB)    Configuration
  0x00010000    0x110000 (1088kB)   nanoCLR
  0x00120000    0x1B0000 (1728kB)   Deployment

Deployment Map
Empty
```

## Description

The [ScanWifi](https://github.com/nanoframework/Samples/tree/master/samples/Wifi/ScanWiFi)-Example does not work with the current libraries. I cloned the repository to my local folder, updated the NuGet packages to the latest stable versions and changed SSID and Password. If I press F5 in VS to deploy and run the sample, it deploys correctly and VS changes into the debug view but quits debugging after a few seconds. The "Output" window shows the error message: 
```
******************************************************************************
** Error: Device stopped after type resolution failure.                     **
** Check in each assembly which assemblies are referenced and their version **
******************************************************************************
The nanoDevice target and the debugger engine failed to initialize because of unspecified device errors.
The debugger engine thread has terminated unexpectedly with error 'Device stopped after type resolution failure. Can't start execution.'.
```

## Detailed repro steps so we can see the same problem

1. Flash the latest stable version to the ESP32 using `nanoff --stable --target ESP32_WROOM_32 --update --serialport COM3` (obviously using the correct port for your environment)
1. Clone the Samples project
1. Open the Wifi demo project in Visual Studio
1. Update NuGet package references to the latest stable version so that type resolution errors in VS disappear and everything (ESP32 firmware and NuGet packages) is on the latest stable version
1. Change SSID and Password in "Program.cs" to match your own settings (I imagine this step can be left out)
1. Press F5 to deploy
1. After deployment, it will change into Debug view for a short period of time and then exit the Debug view with the error message mentioned above in the "Output" window

## Other suggested things

I already tried erasing the device deployment area to make sure that no other libraries were left behind. I also reflashed the ESP32 with the firmware to start from a clean configuration, but that did not work either.

## Expected behaviour

Wifi Sample runs and I have a connection with my wireless network.

### Sample Project
Default [ScanWifi](https://github.com/nanoframework/Samples/tree/master/samples/Wifi/ScanWiFi) project

josesimoes, 2019-10-27:
Most likely a mismatch with the referenced assemblies.

Can you please give it a try using the Preview versions? (both fw and Nugets)

lafe, 2019-10-27:
Using the preview versions of the framework as well as the preview packages from the Azure DevOps Feed (based on https://nanoframework.net/2018/05/16/setup-visual-studio-to-access-preview-versions-feed/), the Wifi works. However, now I get the same error when I am using the MQTT library (nanoFramework.M2Mqtt, Version 4.4.0-alpha.4).

Current capabilities of the device:

```
System Information
HAL build info: 1.4.0.0, nanoFramework running ESP32 built with IDF v3.1
Target: ESP32_WROOM_32
Platform: ESP32

Image build @ Oct 23 2019 GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Solution Build Info: 1.4.0.0, nanoFramework running ESP32 built with IDF v3.1

AppDomains:

Assemblies:
  MQTT, 1.0.0.0
  nanoFramework.Hardware.Esp32, 1.2.0.0
  nanoFramework.Runtime.Events, 1.4.0.0
  Windows.Devices.Wifi, 1.2.0.0
  nanoFramework.Runtime.Native, 1.2.0.0
  mscorlib, 1.4.1.0
  nanoFramework.M2Mqtt, 4.4.0.0
  System.Net, 1.4.0.0

Native Assemblies:
  mscorlib v100.4.1.0, checksum 0x031379FD
  nanoFramework.Runtime.Native v100.0.6.0, checksum 0x035EC7E8
  nanoFramework.Hardware.Esp32 v100.0.7.0, checksum 0xC16385A2
  nanoFramework.Devices.OneWire v100.0.3.3, checksum 0xC7BEAEBC
  nanoFramework.Networking.Sntp v100.0.4.3, checksum 0xB617FDFB
  nanoFramework.ResourceManager v100.0.0.0, checksum 0xCDF9E76C
  nanoFramework.Runtime.Events v100.0.6.0, checksum 0x9E949BB1
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.3, checksum 0xC5A2A195
  System.Net v100.1.3.0, checksum 0x0B718489
  Windows.Devices.Adc v100.1.3.1, checksum 0xA84B23D2
  Windows.Devices.Gpio v100.1.1.0, checksum 0xD7F65BB7
  Windows.Devices.I2c v100.2.0.0, checksum 0x100B5D05
  Windows.Devices.Pwm v100.1.3.1, checksum 0x8E93FB40
  Windows.Devices.SerialCommunication v100.1.0.0, checksum 0x4B336BE9
  Windows.Devices.Spi v100.1.3.0, checksum 0xBDDC51FB
  Windows.Devices.Wifi v100.0.6.0, checksum 0x5AC050C9
  Windows.Storage v100.0.1.0, checksum 0x8D8C8026

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x200000c0  0x00020000
  FLASH 0x08000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x110000     nanoCLR
      1    0x00120000       1      0x1B0000     Deployment
      2    0x002D0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x002D0000    0x040000 (256kB)    Configuration
  0x00010000    0x110000 (1088kB)   nanoCLR
  0x00120000    0x1B0000 (1728kB)   Deployment

Deployment Map
Empty
```

josesimoes, 2019-10-27:
OK. Thanks. That confirms that the root cause is a mismatch in the referenced assemblies. Please wait for a new release with this fixed.

Glad that you've taken the time to report and follow-up this issue. 🙂

josesimoes, 2019-10-30:
Should be fixed with Wi-Fi v1.1 (stable release) and also on latest v1.2.0-preview.4.
MQTT should be OK now. The dependencies update PRs were stuck on the queue… 🙄

closed 2019-10-30 by josesimoes