nanoframework/Home #538: STM32F Discovery boards can't be detected in VS Device Explorer

MarcinBonka, 2019-11-10:
Hello,
My 2 different STM32 discovery boards (F0 or F4) can't be detected and visible in the DeviceExplorer window of Visual Studio under Windows 10. No issues with detecting Nucleo boards. I've noticed when connecting Nucleo board, virtual COM port shows up in the system device manager, Device Explorer searches through available COM ports then finds virtual com port of Nucleo. In case of discovery board, no Virtual com port appears when connected (only STM32 stlink device) so Device Explorer can't find the discovery board. I tried to install newest st-link drivers but no change. Any hints how can I connect/install STM32 discovery boards?

josesimoes, 2019-11-10:
The fact that no COM port shows when you connect your discovery boards is an indication that something is not working with the Stlink.

Try update the Stlink software and the boards firmware. 

We have both F0 and F4 (several) reference targets working so there is no known issue with this. 

Recommend that you go to our Discord community for a more efficient troubleshooting of this situation.

closed 2019-11-10 by josesimoes

MarcinBonka, 2019-11-10:
Thank You for fast response. 
I’ve submitted a question to Discord community.

Wysłane z aplikacji Poczta dla Windows 10

Od: José Simões
Wysłano: 10 November 2019 21:00
Do: nanoframework/Home
DW: MarcinBonka; Author
Temat: Re: [nanoframework/Home] STM32F Discovery boards can't be detected inVS Device Explorer (#538)

The fact that no COM port shows when you connect your discovery boards is an indication that something is not working with the Stlink.
Try update the Stlink software and the boards firmware.
We have both F0 and F4 (several) reference targets working so there is no known issue with this.
Recommend that you go to our Discord community for a more efficient troubleshooting of this situation.