nanoframework/Home #540: OneWire UART TX is configured as active Hi

up-streamer, 2019-11-11:
## Details about Problem
The OneWire bus controller  needs to release the line in order to  allow the host to pull down it during read time slots.  
As per stated in datasheets: "The control line requires a weak pullup resistor since all devices are linked to the bus via a 3-state or open-drain port (the DQ pin in the case of the DS18B20)..." 

**Target:**
ESP-32
**Firmware image version:**
1.4.0-Preview
Worked before? If so, with which nanoFramework image version:
Not checked

## Description

This bug causes the master to held the line hi when the slaves tries to pull it to low level.
This causes abnormal reading values when two slave devices are connected

## Detailed repro steps so we can see the same problem

1. Install two devices on OneWire bus. (tested using DS18B20)

2. Do FindFirst() - FindNext() device ID.

3. Make a loop to read temperature of one device.

4. Touch on the selected device, as the temperature rises it will show inconsistent values.

5. If an oscilocope is available, observe how the bus behave.

...

## Expected behaviour

The OneWire bus should be able to handle more than one device.

## Screenshot

[image]
Waveform showing the Master "fighting" with the slave trying to put zero level on the bus.

## Additional context

josesimoes, 2019-11-12:
According to MAXIM's [documentation](https://www.maximintegrated.com/en/design/technical-documents/tutorials/2/214.html) on UART implementation of 1-Wire master the pins have to be Open Drain capable.

josesimoes, 2020-02-12:
Will this be fixed when addressing #567 ?

closed 2020-02-13 by josesimoes