nanoframework/Home #545: HTTP WebRequest throws SocketException (Possible Bug Reintroduced)

JeremyKnaggs, 2019-11-19:
## Details about Problem

**Target:** ESP32

**Firmware image version:** 1.4.0-Preview.55

Worked before? If so, with which nanoFramework image version:
 Yes, currently works with stable FW 1.3.2

## Description
I'm guessing this is the same bug that was submitted via HTTP WebRequest throws SocketException #542...
HTTP (non ssl) web requests result in exception: exception of type 'System.Net.WebException' occurred in System.Net.Http.dll

## Detailed repro steps so we can see the same problem
Copy of Visual Studio Project which encounters the error: 
[WxStation.zip](https://github.com/nanoframework/Home/files/3861825/WxStation.zip)

1. Using ESP32 and firmware 1.4.0-Preview.55. 

2. Run below method to request text at URL http://knaggsfam.com/hi.txt: 

            public static void Main()
            {
            var networkHerlpers = new NetworkHelpers();
            NetworkHelpers.SetupAndConnectNetwork(true);

            Console.WriteLine("Waiting for network up and IP address...");

            NetworkHelpers.IpAddressAvailable.WaitOne();

            Console.WriteLine("Waiting for valid Date & Time...");

            NetworkHelpers.DateTimeAvailable.WaitOne();
            string url = "http://knaggsfam.com/hi.txt";

            Console.WriteLine($"Performing Http request to: {url}");

            using (var httpWebRequest = (HttpWebRequest)WebRequest.Create(url))
            {
                httpWebRequest.Method = "GET";
                using (var stream = httpWebRequest.GetResponse().GetResponseStream())
                {
                    // read response in chunks of 1k

                    byte[] buffer = new byte[1024];
                    int bytesRead = 0;

                    Console.WriteLine("Http response follows");
                    Console.WriteLine(">>>>>>>>>>>>>");

                    do
                    {
                        bytesRead = stream.Read(buffer, 0, buffer.Length);
                        Console.Write(Encoding.UTF8.GetString(buffer, 0, bytesRead));
                    }
                    while (bytesRead >= buffer.Length);
                }

                Console.WriteLine(">>>>>>>>>>>>>");
                Console.WriteLine("End of Http response");
            }
        }

3. Above method results in output:
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.Net.Sockets.NativeSocket::ioctl [IP: 0000] ++++
    ++++ System.Net.Sockets.Socket::get_Available [IP: 0018] ++++
    ++++ System.Net.Sockets.NetworkStream::get_Length [IP: 001f] ++++
    ++++ System.Net.InputNetworkStreamWrapper::Read_HTTP_Line [IP: 004b] ++++
    ++++ System.Net.HttpWebRequest::ParseHTTPResponse [IP: 002e] ++++
    ++++ System.Net.HttpWebRequest::GetResponse [IP: 0035] ++++
    ++++ WxStation.Program::Main [IP: 0050] ++++
    ++++ Exception System.Net.WebException - 0x00000000 (1) ++++
    ++++ Message: 
    ++++ System.Net.HttpWebRequest::GetResponse [IP: 00c8] ++++
    ++++ WxStation.Program::Main [IP: 0050] ++++
Exception thrown: 'System.Net.WebException' in System.Net.Http.dll
An unhandled exception of type 'System.Net.WebException' occurred in System.Net.Http.dll
...

## Expected behaviour
Current Stable build 1.3.2 results in the expected output (txt file at URL contains 'hi'):

Performing Http request to: http://knaggsfam.com/hi.txt
Http response follows
>>>>>>>>>>>>>
hi>>>>>>>>>>>>>
End of Http response

josesimoes, 2019-11-19:
I've tested this with STM32F769I and it's working as expected, returning that "hi" string.

Possibly something specific to ESP32...

josesimoes, 2019-11-19:
Anyone from @nanoframework/esp32-developers can test this plese?

AdrianSoundy, 2019-11-23:
The exception is being caused by LWIP_SO_RCVBUF not being defined for the build.

closed 2019-11-23 by josesimoes

karayakar, 2019-12-12:
Proper usage;
`using (var httpWebRequest = (HttpWebRequest)WebRequest.Create(url))
            {
                httpWebRequest.Method = "GET";
                var httpResponse = (HttpWebResponse)httpWebRequest.GetResponse();
                using (var stream = httpResponse.GetResponseStream())
                {
                    // read response in chunks of 1k

                    byte[] buffer = new byte[1024];
                    int bytesRead = 0;
                    int streamLength = (int)stream.Length;

                    Console.WriteLine("Http response follows");
                    Console.WriteLine(">>>>>>>>>>>>>");

                    while ((bytesRead = stream.Read(buffer, 0, (buffer.Length + bytesRead < streamLength ? buffer.Length : streamLength - (buffer.Length + bytesRead)))) != 0)
                    {
                        
                        Console.Write(Encoding.UTF8.GetString(buffer, 0, bytesRead));
                    }
                     
                }

                Console.WriteLine(">>>>>>>>>>>>>");
                Console.WriteLine("End of Http response");
            }`

karayakar, 2019-12-12:
Tested on ESP32-WROOM 32
In case if you don't want to miss bytes from the stream (if last chunk size is less than buffer)

            using (var httpWebRequest = (HttpWebRequest)WebRequest.Create(url))
            {
                httpWebRequest.Method = "GET";
                var httpResponse = (HttpWebResponse)httpWebRequest.GetResponse();
                using (var stream = httpResponse.GetResponseStream())
                {
                    // read response in chunks of 1k
                    byte[] buffer = new byte[1024];
                    int bytesRead = 0;
                    int streamLength = (int)stream.Length;
                    Console.WriteLine("Http response follows");
                    Console.WriteLine(">>>>>>>>>>>>>");
                    while ((bytesRead = stream.Read(buffer, 0, (buffer.Length + bytesRead < streamLength ? buffer.Length : streamLength - (buffer.Length + bytesRead)))) != 0)
                    {                      
                        Console.Write(Encoding.UTF8.GetString(buffer, 0, bytesRead));
                    }                   
                }
                Console.WriteLine(">>>>>>>>>>>>>");
                Console.WriteLine("End of Http response");
            }
        }

josesimoes, 2019-12-12:
@karayakar care to submit a PR to the samples repo wiht this improvement?