nanoframework/Home #546: Netduino Plus support

Matthias1172, 2019-11-22:
Hello.
I have an old Netduino Plus which I wanted to program via nanoFramework.
i couldn't find any firmware for this board. Is it actually supported or is there any way to make it running nF?
Thanks in advance for your support.

josesimoes, 2019-11-22:
Hi,

We don't have a "ready to flash" image for this board.

Adding this as feature request in case someone is willing to add the target to the community targets.

piwi1263, 2019-11-22:
@Matthias1172  What Netduino Plus are you refering to, V1 or v2?

networkfusion, 2019-12-19:
@Matthias1172 given there has been no response, I am going to close the issue, however, if it is still a problem feel free to reopen it. Better yet, please discuss in our discord community first.

closed 2019-12-19 by networkfusion