nanoframework/Home #547: Implementation of System.IO.FileStream and System.IO.File Class

mwi-csa, 2019-11-28:
**nanoFramework area:** 
Visual Studio 2017 with the nF VS2017 Extension 1.2.4.28
We use our own hardware board with an STM32F4 microcontroller. The nF for this board we would link to upload soon.

**Is your feature request related to a problem? Please describe.**
Handling with big files respectively if you only want to read some bytes of a file, this is not possible at the moment.

**Describe the solution you'd like**
Implementation of the FileStream class and the File class of the System.IO namespace of the .net core with the following properties and methods:
FileStream:
- CanRead
- CanWrite
- Length
- Position

- Flush
- Dispose
- Close
- Read
- Write
- WriteByte
- ReadByte

File:
- Exists
- Delete
- Create(String path)
- Copy
- Move

**Describe alternatives you've considered**
Implementation of the FileRandomAccessStream would provide mostly the same methods, but due to a lot of existing code of the .net microframework which used the FileStream and due to the fact, that the FileStream is cross plattform usable because its in the .net core we prefer this solution.

**Additional context**
n/a

piwi1263, 2019-11-28:
@josesimoes Can this somehow be separated out. So people have the ability to select a kind of compatibility mode somewhere (or by package or by option choosing in the build phase or whatever idea might come up) to serve both ideas or do you think it is not worth the effort ?

If I have to choose I'd go for the broadest, future saffy compatibility coverage, i.e. the .net core way as described above.

In the case of only wanting one file handling package I suggest to declare the current package/solution for deprecated as soon as the new one goes live and give people the change to exchange their implementation using the old package vs the new package.

josesimoes, 2019-11-28:
@piwi1263 this is 95% "cosmetics". 
Meaning that the API is quite similar and the native code to make it happen would be pretty much the same,

Agreed that supporting both looks a bit too much work for what's worth.
And, since we are in the early stages of File system, I believe any breaking change (like moving to the .NET Core API) wouldn't cause any serious pains for anyone already using it.

closed 2020-02-03 by stale[bot]