nanoframework/Home #549: Refactoring System.Collections library to work with Unit tests

adaslesniak, 2019-12-13:
After updating nanoframework libraries from 1.4 to 1.6 and moving parts of corelib to other libraries (to make corelib smaller and more friendly for weak devices) calls to classes moved to those libraries can't be used in unit tests. So code that is using functionality defined in dot net core library is not unit testable (same goes if one want to make simulator)

Unit tests cant contain dot net core code, since some core classes are now in different libraries than in dot net. This results in calling nanoframework.system.collections library to execute stack.Push() method and in the end this results in call to external(native) method that is not available in unit tests.

Those classes that rely on native implementation are mainly Stack and Queue - they are all variants of ArrayList which is variant of Array. Those classes can be easily refactored to not use native code, so they will work as shared library (pure c# code executable on multiple machines) Performance penalty (if at all) would be minuscule, while amount of native code would be reduced (Array/Stack/Queue are heavily duplicated ideas of the same).

Expected results:  
1 - library sharable among multiple platforms, no problem in unit testing/simulator code that uses those methods; And since this is functionally part of .Net core -  that is quite big thing.
2 - smaller amount of c++ code required (no need for implementation of those system.collections methods); 
3 - less bugs since there will be less duplicated code (yeah, this code is duplicated in microsoft dot net as well, but need not to be); 
4 - small downside could be that those methods could work slower, but IMHO this would be very tiny difference, detectable maybe in specialized measurements, about 2-5% slower maybe it would be...

Optionally: could also implement LinkedList -  it's much faster for many use cases, where we just collect some elements and then process them once. In case of Array/ArrayList, those tables are copied multiple times when array needs to be expanded and this takes time - also GC time is required to clean up all those too small arrays.

josesimoes, 2019-12-13:
Comments are welcome! 
There is an ongoing conversation in [Discord](https://discord.gg/gCyBu8T) (#class-libraries channel) about this and how it can connect with Unit Testing.

steffalk, 2019-12-13:
I wasn't aware that collections depend on native code at all and I find the idea of implementing it all in managed code fine. Just to have arguments for folks who may point to "issue" 4 (performance), I would suggest doing a small benchmark with the current version and the same with a managed one. As a base, I would vote for cleaner code vs. small performance advantages, but hardcore-µC-devs might disagree ;-)

josesimoes, 2019-12-16:
@adaslesniak the proposed changes (in what concerns the general improvements and code optimization) seem to be consensual. The globaly expressed concern is with the performance impact. 
Which is is perfectly understandable and reasonable. 

I would say that you're good to go with this, conditioned to the performance tests.
Suggest that you use the existing ones from .NETMF. These seem to have enough coverage to provide a good insight on the performance before and after the changes.

closed 2020-02-21 by stale[bot]