nanoframework/Home #550: How can i run DFPlayer with nanoframework?

DenisMtfl, 2019-12-16:
Is it possible?

josesimoes, 2019-12-16:
Please enlighten us on what is DFPLayer...

DenisMtfl, 2019-12-16:
The DFPlayer Mini MP3 Player For Arduino is a small and low price MP3 module with an simplified output directly to the speaker. The module can be used as a stand alone module with attached battery, speaker and push buttons or used in combination with an Arduino UNO or any other with RX/TX capabilities.

Source:
https://wiki.dfrobot.com/DFPlayer_Mini_SKU_DFR0299

DenisMtfl, 2019-12-17:
> Please enlighten us on what is DFPLayer...

@josesimoes 
The DFPlayer Mini MP3 Player For Arduino is a small and low price MP3 module with an simplified output directly to the speaker. The module can be used as a stand alone module with attached battery, speaker and push buttons or used in combination with an Arduino UNO or any other with RX/TX capabilities.

Source:
https://wiki.dfrobot.com/DFPlayer_Mini_SKU_DFR0299

josesimoes, 2019-12-17:
Being controlled over a serial port you're good to go with nanoFramework.
Please check the SerialCommunication pack in the samples repo. You'll find there all you need to get you started.

closed 2019-12-17 by josesimoes

DenisMtfl, 2019-12-17:
Thanks! I will try.

VitalyV, 2022-05-25:
Hi @DenisMtfl.
Did you manage to send a command or read the data from DFPlayer?
Thanks

kamszyc, 2025-05-26:
It can be controlled using Yx5300 class, described here
https://docs.nanoframework.net/devicesdetails/Yx5300/README.html

This issue comes up first on Google search, so I thought this info can help someone