nanoframework/Home #552: Array are only partially initialized

Karlfried, 2019-12-27:
Arrays are only partially initialized if too many constants are specified.

Sample Project:

`using System;
using System.Threading;

namespace NFApp1
{
    public class Program
    {
        public static void Main()
        {
            PixelFont7X9 _font1;

            Console.WriteLine("Hello world!");

            _font1 = new PixelFont7X9(); // font 1

            if (_font1._charmap[80] == 0)
                Console.WriteLine("Wrong, initialization incorrect!!!");

            Thread.Sleep(Timeout.Infinite);
        }

        /// <summary>
        /// Represents a 7 X 9 (W x H) pixel font object to be used with nanoFramework display device drivers.
        /// </summary>
        public class PixelFont7X9
        {
            public UInt32[] _charmap;

            public PixelFont7X9()
            {
                _charmap = new uint[]

                {
                    0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000,
                    // !
                    0x10000000, 0x10000000, 0x10000000, 0x10000000, 0x10000000, 0x10000000, 0x00000000, 0x10000000, 0x10000000,
                    // "
                    0x28000000, 0x28000000, 0x28000000, 0x28000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000,
                    // #
                    0x28000000, 0x28000000, 0x28000000, 0xFE000000, 0x28000000, 0xFE000000, 0x28000000, 0x28000000, 0x28000000,
                    // $
                    0x10000000, 0x38000000, 0x54000000, 0x50000000, 0x38000000, 0x14000000, 0x54000000, 0x38000000, 0x10000000,
                    // %
                    0xC2000000, 0xC6000000, 0x0C000000, 0x18000000, 0x30000000, 0x60000000, 0xC0000000, 0x86000000, 0x06000000,
                    // &
                    0x30000000, 0x48000000, 0x28000000, 0x10000000, 0x28000000, 0x46000000, 0x86000000, 0x88000000, 0x70000000,
                    // '
                    0x10000000, 0x10000000, 0x10000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 0x00000000,
                    // (
                    0x18000000, 0x20000000, 0x20000000, 0x20000000, 0x20000000, 0x20000000, 0x20000000, 0x20000000, 0x18000000
                };
            }
        }
    }
}`

AdrianSoundy, 2020-02-01:
Only the first 16 entries seem to get initialized.

SandorDobos, 2020-04-12:
I have the same issue with an other code (@DWeaver's OLED driver having zeroes in its Font map byte array after the 250th index till the last 1275th).

I made some tests:
- exe, bin and pe files have the full map inside (with good values)
- after reading back my esp32's contents with esptool.py read_flash it is having the whole map inside too.
- during runtime (VS debugging session) I see only the first 250 bytes at various points of the code.
- if I change the array type from byte to int (and place one necessary cast at the only place where the code is using (reading) the array the last good index becomes 58.

Any suggestion where the issue may be lurking may resulting in grabbing this one by me :)
Regards!

valoni, 2020-04-12:
dynamic array fail - static array work

example is as i posted to understand 

```csharp

using System;
using System.Threading;

namespace NFArrays
{
    public class Program
    {
        public static void Main()
        {
            /* TRICK 1*/
            // nor in way     int[] MYlistoff = new int[] {
            int[] MYlistoff = {
                248, 248, 248, 248, 248, 248, 248, 248, 248, 248, 248, 248, 248, 248, 248,//15 
                248, 248, 248, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120,//30
                120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120,//45
                120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120,//60 
                120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120,//75 
                120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120, 120,//90
                248, 248, 248, 120, 120, 120, 120, 120, 250, 200
               };

            Console.WriteLine("DO CHECK !");  //do not work 

            Console.WriteLine("array lenght is " + MYlistoff.Length);
            for (int i = 0; i < MYlistoff.Length; i++)
            {
                Console.WriteLine("i(" + i.ToString()+" )= " + MYlistoff[i].ToString());
                Thread.Sleep(1);
            }

            /* TRICK 2*/
            int[] MYlistoff2 = new int[100];

            MYlistoff2[0] = 21;
            MYlistoff2[1] = 12;
            MYlistoff2[2] = 222;
            MYlistoff2[3] = 32;
            MYlistoff2[4] = 23;
            MYlistoff2[5] = 42;
            MYlistoff2[6] = 24;
            MYlistoff2[7] = 52;
            MYlistoff2[8] = 25;
            MYlistoff2[9] = 62;
            MYlistoff2[10] = 26;
            MYlistoff2[11] = 72;
            MYlistoff2[12] = 26;
            MYlistoff2[13] = 25;
            MYlistoff2[14] = 24;
            MYlistoff2[15] = 23;
            MYlistoff2[16] = 22;
            MYlistoff2[17] = 233;
            MYlistoff2[18] = 233;
            MYlistoff2[19] = 24;
            MYlistoff2[20] = 24;
            MYlistoff2[21] = 24;
            MYlistoff2[22] = 255;
            MYlistoff2[23] = 255;
            MYlistoff2[24] = 255;
            MYlistoff2[25] = 52;
            MYlistoff2[26] = 233;
            MYlistoff2[27] = 244;
            MYlistoff2[28] = 255;
            MYlistoff2[29] = 552;
            MYlistoff2[30] = 23;
            MYlistoff2[31] = 23;
            MYlistoff2[32] = 23;
            MYlistoff2[33] = 23;
            MYlistoff2[34] = 23;
            MYlistoff2[35] = 23;
            MYlistoff2[36] = 23;
            MYlistoff2[37] = 23;
            MYlistoff2[38] = 23;
            MYlistoff2[39] = 23;
            MYlistoff2[40] = 22;
            MYlistoff2[41] = 23;
            MYlistoff2[42] = 24;
            MYlistoff2[43] = 25;
            MYlistoff2[44] = 266;
            MYlistoff2[45] = 772;
            MYlistoff2[46] = 288;
            MYlistoff2[47] = 992;
            MYlistoff2[48] = 299;
            MYlistoff2[49] = 2;
            MYlistoff2[50] = 222;
            MYlistoff2[51] = 221;
            MYlistoff2[52] = 12;
            MYlistoff2[53] = 2;
            MYlistoff2[54] = 22;
            MYlistoff2[55] = 22;
            MYlistoff2[56] = 22;
            MYlistoff2[57] = 22;
            MYlistoff2[58] = 22;
            MYlistoff2[59] = 22;
            MYlistoff2[60] = 22;
            MYlistoff2[61] = 22;
            MYlistoff2[62] = 22;
            MYlistoff2[63] = 222;
            MYlistoff2[64] = 22;
            MYlistoff2[65] = 22;
            MYlistoff2[66] = 22;
            MYlistoff2[67] = 22;
            MYlistoff2[68] = 22;
            MYlistoff2[69] = 22;
            MYlistoff2[70] = 2;
            MYlistoff2[71] = 2;
            MYlistoff2[72] = 2;
            MYlistoff2[73] = 2;
            MYlistoff2[74] = 2;
            MYlistoff2[75] = 2;
            MYlistoff2[76] = 2;
            MYlistoff2[77] = 2;
            MYlistoff2[78] = 2;
            MYlistoff2[79] = 2;
            MYlistoff2[80] = 2;
            MYlistoff2[81] = 2;
            MYlistoff2[82] = 2;
            MYlistoff2[83] = 2;
            MYlistoff2[84] = 2;
            MYlistoff2[85] = 2;
            MYlistoff2[86] = 2;
            MYlistoff2[87] = 2;
            MYlistoff2[88] = 2;
            MYlistoff2[89] = 2;
            MYlistoff2[90] = 2;
            MYlistoff2[91] = 2;
            MYlistoff2[92] = 2;
            MYlistoff2[93] = 2;
            MYlistoff2[94] = 2;
            MYlistoff2[95] = 2;
            MYlistoff2[96] = 2;
            MYlistoff2[97] = 2;
            MYlistoff2[98] = 2;
            MYlistoff2[99] = 2;
            MYlistoff2[90] = 2;

            Console.WriteLine("DO CHECK 2!"); //work

            Console.WriteLine("array lenght is " + MYlistoff2.Length);
            for (int j = 0; j < MYlistoff2.Length; j++)
            {
                Console.WriteLine("j(" + j.ToString() + " )= " + MYlistoff2[j].ToString());
                Thread.Sleep(1);
            }

            Thread.Sleep(Timeout.Infinite);
        }
    }
}

```

SandorDobos, 2020-04-12:
I tracked down till RuntimeHelpers:InitializeArray.
This lines: 

```
CLR_PMETADATA ptrSrc  = inst.m_assm->GetSignature( inst.m_target->defaultValue );
CLR_UINT32    lenSrc; NANOCLR_READ_UNALIGNED_UINT16( lenSrc, ptrSrc );
```
lenSrc becomes 250 in my case, which is 1111 1011
the real size in my case is 1275, which is 0100 1111 1011
first bytes of ptrSrc are 0xfb 0x0 0x0 0x0
so the newly created and tobeinitialized array looks bad
it is losting that 0100 from its size specs

josesimoes, 2020-04-13:
@SandorDobos thanks for digging into this. Seems that's something with the signature that's not properly storing the length value.

closed 2020-04-13 by josesimoes