nanoframework/Home #553: vsx "GenerateBinaryOutputTask" task failed unexpectedly

TheTrigger, 2020-01-03:
## Details about Problem

**nanoFramework area:** Visual Studio extension

**VS version:** 15.9.18 / 16.5.0-preview1

**VS extension version:** 1.3.0.1

## Description

While building, binaries will be created, but the compiler gives an error for each project:
```
Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB4018	The "GenerateBinaryOutputTask" task failed unexpectedly.
System.IO.IOException: The process cannot access the file 'C:\Users\Fabio\source\repos\NFApp1\NFClassLibrary1\bin\Debug\NFClassLibrary1.dll' because it is being used by another process.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.FileStream.Init(String path, FileMode mode, FileAccess access, Int32 rights, Boolean useRights, FileShare share, Int32 bufferSize, FileOptions options, SECURITY_ATTRIBUTES secAttrs, String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize, FileOptions options, String msgPath, Boolean bFromProxy)
   at System.IO.FileStream..ctor(String path, FileMode mode)
   at nanoFramework.Tools.GenerateBinaryOutputTask.Execute()
   at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute()
   at Microsoft.Build.BackEnd.TaskBuilder.<ExecuteInstantiatedTask>d__26.MoveNext()	NFClassLibrary1	C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\nanoFramework\v1.0\NFProjectSystem.MDP.targets	718	
```

## Detailed repro steps to reproduce

1. Create new exe project

2. Add a library

3. Build

## Screenshot

[image]

[image]

[image]

thopiekar, 2020-01-30:
Is there a known workaround for this problem? How is the status on fixing this issue? 😃

TheTrigger, 2020-01-30:
> Is there a known workaround for this problem? How is the status on fixing this issue? 😃

Try a downgrade, but I don't know where to find it...

josesimoes, 2020-01-30:
@thopiekar the only workaround is to downgrade to an older version.

As for the fixing: it's on the queue. Work on nanoFramework is done mostly in free time.

thopiekar, 2020-01-30:
Alright. I was looking for the previous version before and I was wondering whether other versions of the VS extension are missing, too..
Good that it has been clearified! 👍  - https://github.com/nanoframework/Home/issues/558

josesimoes, 2020-02-06:
This is something that seems to be plaguing others too. Related with the Task not being shutdown after completion to allow reusing it.

dotnet/runtime#19961
microsoft/msbuild#3141
https://github.com/microsoft/msbuild/issues/2916#issuecomment-360897413

closed 2020-02-06 by josesimoes