nanoframework/Home #555: DFU packages only reach nanoBooter if memory not fully cleared

networkfusion, 2020-01-17:
## Details about Problem

**nanoFramework area:** 
nanoFirmwareFlasher 
ST DFU Demo

**Target*
OrgPalThree (and maybe others)

**Firmware image version:**
Latest (as of 2019-01-17)

## Description

When flashing a DFU package to a board that already contains the booter and CLR, once updated successfully, the board only boots into nanoBooter, and seems incapable of reaching the CLR

## Workaround
Use the ST DFU testing app to fully erase the memory - Side effect is that all config region is lost.

## Possible fixes
Only create a DFU file that contains the CLR
Move Config Region to end of memory
...

networkfusion, 2020-01-20:
Further ideas / resources at:
https://github.com/networkfusion/St-Dfu
https://github.com/pimoroni/DfuSE/

networkfusion, 2020-03-03:
@piwi1263 does the erase work as expected with the latest nanoff release?

networkfusion, 2020-04-19:
Still seems to be an issue with the orgpal3 after the recent improvements to nanoff.

networkfusion, 2020-05-24:
Tested as of 2020-05-24 with latest image and nanoff, but is still an issue.

josesimoes, 2020-05-28:
@networkfusion what is the version of nanoff that you're using?
I've just run this with the latest (v1.12.1) and also on VS in debug and the update is performed as expected. I've checked the config block and the contents remain untouched during the udpate.

closed 2020-05-28 by josesimoes