nanoframework/Home #556: FileIO.ReadText only returns first line of text

networkfusion, 2020-01-21:
## Details about Problem
When reading a text file using `FileIO.ReadText(file)` only the first line is returned.

This is because of the native function `static THD_FUNCTION(ReadTextWorkingThread, arg)` that uses f_gets which returns on a new line char `n`

The fix would be to use a while loop until the NULL char `\0` delimiting the end of file is received.

networkfusion, 2020-05-14:
Since the workaround is pretty simple and there is no guarantee on the encoding used (unless we support the optional `public static IAsyncOperation<string> ReadTextAsync(IStorageFile file, UnicodeEncoding encoding);` from UWP, should this function be marked as unsupported, or should it be solved?

networkfusion, 2021-09-10:
Given this lib is now deprecated, It needs re-evaluating to ensure it is still an issue. If not, I suggest it is closed.

loanst, 2022-01-03:
I'm met strange error when tried to get data from file not from the first line using FileSystem (from internal memory)

This code was used:
using (FileStream fileStream = new FileStream(dayFilePath, FileMode.Open, FileAccess.Read))
            {
                byte[] fileContent = new byte[100];
                fileStream.Read(fileContent, 50, 100);
                string dataFromFile = Encoding.UTF8.GetString(fileContent, 0, fileContent.Length);

                return dataFromFile;
            }

and this threw an error:

    ++++ Exception System.IndexOutOfRangeException - 0x00000000 (5) ++++
    ++++ Message: Buffer size is smaller then offset + byteCount.
    ++++ System.IO.FileStream::Read [IP: 002f] ++++
    ++++ MqSensorsTest.Program::ReadFromFile [IP: 003a] ++++
    ++++ MqSensorsTest.ControllerTest::GetSensorsData [IP: 00a7] ++++
    ++++ System.Reflection.MethodBase::Invoke [IP: 0000] ++++
    ++++ nanoFramework.WebServer.WebServer+<>c__DisplayClass45_0::<StartListener>b__0 [IP: 0185] ++++
Exception thrown: 'System.IndexOutOfRangeException' in System.IO.FileSystem.dll
An unhandled exception of type 'System.IndexOutOfRangeException' occurred in System.IO.FileSystem.dll
Additional information: Buffer size is smaller then offset + byteCount.
[image]

Please, have a look on screen

loanst, 2022-01-04:
This code works for me:
            string dataFromFile = "";
            var file = StorageFile.GetFileFromPath(dayFilePath);
            var readBuffer = FileIO.ReadBuffer(file);
            var readBufferLength = readBuffer.Length;
            var lastLineNumber = startLine + linesCount;
            var lineSize = (uint)GetLineLenght();
            using (DataReader dataReader = DataReader.FromBuffer(readBuffer))
            {
                for (int i = 0; i < lastLineNumber; i++)
                {
                    if ((i * lineSize) >= readBufferLength)
                        return dataFromFile;

                    var readValue = dataReader.ReadString(lineSize);
                    if (i >= startLine)
                    {
                        dataFromFile += readValue;
                    }
                }
            }

            return dataFromFile; 
            
            
For this type of data in the text file:
2022-01-04T22:15:32 4095 3033 3618
2022-01-04T22:15:37 4095 3037 3629
2022-01-04T22:15:42 4095 3037 3607
2022-01-04T22:15:47 4095 3037 3607
2022-01-04T22:15:52 4095 3038 3606

closed 2022-05-11 by networkfusion