nanoframework/Home #557: nf_utils not found

racerxdl, 2020-01-26:
## Details about Problem

**nanoFramework area: Other Tools**

**VS version: Code 1.41.1**

**Target: ESP32**

## Description

Followed the ESP32 tutorial but cmake cant find `nf_utils`:

```
[cmake] Platform ESP32 target board: ESP32_WROOM_32
[cmake] CMake Error at targets/FreeRTOS_ESP32/CMakeLists.txt:6 (include):
[cmake]   include could not find load file:
[cmake] 
[cmake]     nf_utils
[cmake] 
[cmake] 
```

## Detailed repro steps so we can see the same problem

Just follow http://docs.nanoframework.net/content/getting-started-guides/build-esp32.html#Manual-Install-of-the-build-environment-for-ESP32 and hit `F7`
...

## Other suggested things

## Expected behaviour

A clear and concise description of what you expected to happen.

## Screenshot

[image]

## Additional context

Ubuntu 18.04

Tried joining discord, but the invite is outdated.

josesimoes, 2020-01-27:
`nf_utils` it's a CMake module and it's located at CMake\Modules\NF_Utils.cmake.
As you seem to be building this on a *nix machine it strikes me that this could be related with the file name. Can you test this: rename it to all lower case and give it another try?

BTW, although building in *nix machines should be perfectly all right know that there is no support for C# debugging on these machines. Only Windows 10.

josesimoes, 2020-01-27:
> Tried joining discord, but the invite is outdated.

This is the official invite URL: https://discord.gg/gCyBu8T
Is this the one you tried? I've double checked it and it shows as active and seems to be working...

racerxdl, 2020-02-02:
Hey @josesimoes, looks like there is no NF_Utils.cmake anywhere. https://github.com/nanoframework/nf-interpreter/tree/develop/CMake

The link of the discord worked now. Weird.

josesimoes, 2020-02-02:
https://github.com/nanoframework/nf-interpreter/blob/develop/CMake/Modules/NF_Utils.cmake

racerxdl, 2020-02-03:
So I just tested, looks like the nf_utils wasn't really a problem. I changed

```
include(nf_utils)
```

to 

```
include(NF_Utils)
```

In `targets/FreeRTOS_ESP32/CMakeLists.txt`. This error disappeared:

```
[cmake] CMake Error at targets/FreeRTOS_ESP32/CMakeLists.txt:6 (include):
[cmake]   include could not find load file:
[cmake] 
[cmake]     nf_utils
```

But there is still the one from the image

```
[cmake] Wire Protocol TRACE_MASK is '0'
[cmake] CMake Error at targets/FreeRTOS_ESP32/ESP32_WROOM_32/CMakeLists.txt:35 (find_package):
[cmake]   By not providing "FindNF_NETWORKING.cmake" in CMAKE_MODULE_PATH this
[cmake]   project has asked CMake to find a package configuration file provided by
[cmake]   "NF_NETWORKING", but CMake did not find one.
[cmake] 
[cmake]   Could not find a package configuration file provided by "NF_NETWORKING"
[cmake]   with any of the following names:
[cmake] 
[cmake]     NF_NETWORKINGConfig.cmake
[cmake]     nf_networking-config.cmake
[cmake] 
[cmake]   Add the installation prefix of "NF_NETWORKING" to CMAKE_PREFIX_PATH or set
[cmake]   "NF_NETWORKING_DIR" to a directory containing one of the above files.  If
[cmake]   "NF_NETWORKING" provides a separate development package or SDK, be sure it
[cmake]   has been installed.
[cmake] 
```

josesimoes, 2020-02-03:
That new error looks pretty much the same as the previous one. This is clearly an issue with file name being case sensitive on *nix systems.

If you could go through all these and submit a PR with the fixes that would be awesome! 😃

racerxdl, 2020-02-03:
I can fix that. But I can't find this `NF_Networking`:

```
┌─[lucas@nblucas] - [/media/ELTN/Works/nf-interpreter] - [Mon Feb 03, 12:55]
└─[$] <git:(develop*)> find . -iname \*network\*
./CMake/Modules/FindnanoFramework.Networking.Sntp.cmake
./CMake/Modules/FindWindows.Networking.Sockets.cmake
./CMake/Modules/FindNF_Networking.cmake
./targets/FreeRTOS_ESP32/ESP32_WROOM_32/nanoCLR/Windows.Networking.Sockets
./targets/FreeRTOS_ESP32/ESP32_WROOM_32/nanoCLR/Windows.Networking.Sockets/win_net_sockets_native_Windows_Networking_Sockets_SocketNative.cpp
./targets/FreeRTOS_ESP32/ESP32_WROOM_32/Network
./targets/FreeRTOS_ESP32/ESP32_WROOM_32/Network/Target_Network.cpp
./targets/FreeRTOS_ESP32/ESP32_WROOM_32/Network/targetHAL_Network.cpp
./targets/CMSIS-OS/ChibiOS/nanoCLR/Windows.Networking.Sockets
./targets/CMSIS-OS/ChibiOS/nanoCLR/Windows.Networking.Sockets/win_net_sockets_native_Windows_Networking_Sockets_SocketNative.cpp
./targets/CMSIS-OS/ChibiOS/common/Target_Network.cpp
./targets/CMSIS-OS/ChibiOS/common/targetHAL_Network.cpp
./targets/FreeRTOS/NXP/nanoCLR/Target_Network.cpp
./targets/FreeRTOS/NXP/nanoCLR/targetHAL_Network.cpp
./src/Windows.Networking.Sockets
./src/PAL/nanoPAL_Network_stubs.cpp
./src/PAL/Include/nanoPAL_Network.h
./src/HAL/Include/nanoHAL_Network.h
./src/DeviceInterfaces/Networking.Sntp
./src/DeviceInterfaces/Networking.Sntp/nf_networking_sntp.cpp
./src/DeviceInterfaces/Networking.Sntp/nf_networking_sntp_nanoFramework_Networking_Sntp.cpp
./src/DeviceInterfaces/Networking.Sntp/nf_networking_sntp.h
./src/DeviceInterfaces/System.Net/sys_net_native_System_Net_NetworkInformation_Wireless80211Configuration.cpp
./src/DeviceInterfaces/System.Net/sys_net_native_System_Net_NetworkInformation_NetworkInterface.cpp
./src/DeviceInterfaces/System.Net/sys_net_native_System_Net_NetworkInformation_WirelessAPConfiguration.cpp
```

Not even here in github:

https://github.com/nanoframework/nf-interpreter/search?q=NF_NETWORKINGConfig&unscoped_q=NF_NETWORKINGConfig

Is it in a different repository?

josesimoes, 2020-02-04:
@racerxdl `NF_Networking` is `FindNF_Networking.cmake` that's a CMake convention for the name of the components. 

Know that this is building daily by Azure Pipelines and everything that needs to be there it is. Just that's a Windows machine.

racerxdl, 2020-02-06:
Ok, so I will take a deep look into cmake files and see if I can get it building on linux, then I will make a Pull Request ^^

josesimoes, 2020-02-12:
hey @racerxdl any update on this? 😁

networkfusion, 2020-03-03:
@racerxdl , have you managed to solve this specific issue? If so, Perhaps this issue can be closed and the linux support case can be discussed on discord, or at least the title of the issue updated for it's TRUE intent...

racerxdl, 2020-03-03:
Hi @networkfusion  / @josesimoes  sorry for the delay. I haven't managed to fix all the issues yet, but they seen to be all related to case-sensitive issues. I'm thinking about trying to fix it and add a case-sensitive check to the Build Pipeline, so even if someone develops in a non case-sensitive OS (like Windows) it breaks the build when opening a PR. Then everyone can fix right away.

Still, there is LOTs of file with wrong naming, so I didn't got time to fix all of them. But I will keep doing that.

josesimoes, 2020-04-29:
hello @racerxdl here I am asking for your update on this. Feel free to submit a PR with what you have fixed. Someone else can pick up where you left and fix the rest.

closed 2020-07-05 by stale[bot]

josesimoes, 2021-02-08:
Duplicate of #692.
Fixed there.