nanoframework/Home #558: Extensions: Missing version tags and install packages

thopiekar, 2020-01-30:
## Description

I would like to test different versions of the extension to be able to trace errors in the extension itself. Otherwise, understanding new bugs is difficult and not traceable.

## Detailed repro steps so we can see the same problem

See: https://github.com/nanoframework/nf-Visual-Studio-extension/releases

## Expected behavior

Every released version of the VS extension is tagged and includes an install package as done in the past.

## Screenshot

## Additional context

Add any other context about the problem here.

josesimoes, 2020-01-30:
The only tags that were missing are the ones for v1.3.0.1.
Published now.

closed 2020-01-30 by josesimoes

thopiekar, 2020-01-30:
Many thanks @josesimoes