nanoframework/Home #559: Adding new Resource file

valoni, 2020-01-30:
Please read the following information before posting the issue.

## Details about Problem

when you try to add an resource file on Visual Studio 2019 it stuck forever on loop 

Visual Studio 2019 v16.4.3 vsix extension 1.3.0.1 

## Description

A clear and concise description of what the bug is.

## Detailed repro steps so we can see the same problem

step to reproduce that 

[image]

[image]

josesimoes, 2020-01-30:
Could this be a regression of #470? Or maybe they have "fixed" something in the standard targets files that is now colliding with the previous fix...

josesimoes, 2020-01-30:
Does this happen in VS2017 too?

valoni, 2020-01-30:
> Does this happen in VS2017 too?

on Visual Studio 2017 work well - no problem or any issues with Resources seem only on Visual Studio 2019 cause a break.

josesimoes, 2020-01-31:
Oh... here we go again… 🤕

josesimoes, 2020-03-20:
Reported to VS Developer Community. See [here](https://developercommunity.visualstudio.com/content/problem/956161/resource-file-with-visual-studio-2019-v165.html?childToView=956382#comment-956382)

## Please upvote the issue above so it get's higher on the priority list

mariodivece, 2020-07-13:
Same issue here with VS 2019 :(

josesimoes, 2020-07-13:
@mariodivece yes the issue remains unsolved. It's stuck on Microsoft queue.
It will get attention when enough people up vote.
Please go to the VS developer community site link above and up vote there.

meanshe, 2020-07-19:
Same issue here with VS 2019 :(

josesimoes, 2020-07-19:
@meanshe the issue remains to be solved by Microsoft. Please up vote on the link above.

josesimoes, 2020-08-06:
Issue fixed. Schedule to release with VS 16.8.

valoni, 2020-10-07:
on wait from MS to resolve problem ..

valoni, 2020-11-10:
Resolved with latest update of Visual Studio 16.8.0

closed 2020-11-10 by valoni

antonsterkhov, 2022-01-28:
Same issue here with VS 2022 :(