nanoframework/Home #560: Implement more flexible SPI pin definitions

gojimmypi, 2020-01-30:
**nanoFramework area:** Hardware/target board
This suggestion  specifically mentions the ESP32, but would likely be useful on all hardware platforms.

**Is your feature request related to a problem? Please describe.**
The current ESP32 SPI implementation does not work on the ULX3S ESP32-WROOM as some pin definitions are hardcoded in the hardware abstraction firmware layer for HSPI and VSPI 

**Describe the solution you'd like**
It would be great to be able to pass a full definition of *all* SPI pins used in a C# application to the underlying hardware abstraction layer. (yes, one could then argue if that is actually a HAL or not) ;)

I'm willing to help work on this if there is agreement on architecture and implementation. 

**Describe alternatives you've considered**
An alternative is to do "soft" SPI at the C# level. This is of course, considerably slower. See [my OLED_SSD1331 example](https://github.com/gojimmypi/ulx3s-examples/blob/SSD1331/nanoFramework/nanoFrameworkSSD1331/OLED_SSD1331.cs#L49).

**Additional context**
This topic was discussed briefly on the discord channel.

I'm working on getting the [ULX3S](https://radiona.org/ulx3s/) SSD1331 display driver working with the nanoFramework library in my [ulx3s-examples SSD1331 branch](https://github.com/gojimmypi/ulx3s-examples/tree/SSD1331/nanoFramework/nanoFrameworkSSD1331_demo).

My first attempt at using the built-in libraries resulted in a blank display under all conditions. Slapping on a logic analyzer and sure enough - zero activity on the SPI bus.

I've had some experience with the SPI display on ULX3S in https://github.com/emard/ulx3s/issues/8.  I documented what I learned in my [Visual Micro SPI](https://github.com/gojimmypi/ulx3s-examples/blob/master/VisualMicro-SSD1331-Display/VisualMicro-SSD1331-Display.ino#L88) comments. In particular:

```
// VSPI see https://github.com/espressif/arduino-esp32/blob/7d7824701fe5e22f08555d3e1ce3180a922b2151/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino#L31
// initialise vspi with default pins SCLK = 18, MISO = 19, MOSI = 23, SS = 5
// working WROOM-32 using VSPI (does not work on ULX3S)
//#define oled_csn  5 //  white aka cs - chip select
//#define oled_dc   26 // orange aka ds aka a0 -  SPI data or command selector pin
//#define oled_resn 27 // gray aka rst - reset
//#define oled_mosi 23 // purple aka mosi - data
//#define oled_clk  18 // blue aka sclk - clock
//#define oled_miso -1 // not used

// HSPI see https://github.com/espressif/arduino-esp32/blob/7d7824701fe5e22f08555d3e1ce3180a922b2151/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino#L37
// initialise hspi with default pins SCLK = 14, MISO = 12, MOSI = 13, SS = 15
// working WROOM-32 using HSPI (does not work on ULX3S)
//#define oled_csn  15 // white aka cs - chip select
//#define oled_dc   26 // orange aka ds aka a0 -  SPI data or command selector pin
//#define oled_resn 27 // gray aka rst - reset
//#define oled_mosi 13 // purple aka mosi - data
//#define oled_clk  14 // blue aka sclk - clock
//#define oled_miso -1 // 12 not used

// working ULX3S  
#define oled_csn  17 // aka cs - chip select
#define oled_dc   16 // aka ds aka a0 -  SPI data or command selector pin
#define oled_resn 25 // aka rst - reset
#define oled_mosi 15 // aka mosi - data
#define oled_clk  14 // aka sclk - clock
#define oled_miso -1 // 12 not used
```
Note how neither the HSPI nor VSPI are available. Yes, the ESP32 is sitting behind an FPGA, but that's another story also discussed in https://github.com/emard/ulx3s/issues/8 .

When the nanoFramework calls the SPI driver, only some of the pin definitions are currently assignable by the user. The rest are off in the extern black hole when calling [NativeInit](https://github.com/nanoframework/lib-Windows.Devices.Spi/blob/develop/source/Windows.Devices.Spi/SpiDevice.cs#L64).
[image]

Digging a bit further, we see in the comments that [only VSPI and HSPI are implemented](https://github.com/nanoframework/nf-interpreter/blob/develop/targets/FreeRTOS_ESP32/ESP32_WROOM_32/nanoCLR/Windows.Devices.Spi/win_dev_spi_native_Windows_Devices_Spi_SpiDevice.cpp#L9). There's no room for any other SPI definitions.

Note how the [adafruit drivers](https://github.com/adafruit/Adafruit-SSD1331-OLED-Driver-Library-for-Arduino/blob/master/Adafruit_SSD1331.cpp#L135) allow for passing of all pin definitions:

```
/**************************************************************************/
/*!
    @brief  Instantiate Adafruit SSD1331 driver with software SPI
    @param    cs    Chip select pin #
    @param    dc    Data/Command pin #
    @param    mosi  SPI MOSI pin #
    @param    sclk  SPI Clock pin #
    @param    rst   Reset pin # (optional, pass -1 if unused)
*/
/**************************************************************************/
Adafruit_SSD1331::Adafruit_SSD1331(int8_t cs, int8_t dc, int8_t mosi, int8_t sclk, int8_t rst) :  Adafruit_SPITFT(TFTWIDTH, TFTHEIGHT, cs, dc, mosi, sclk, rst, -1) {
}

```

It would be great to pass pin programmer-selectable numbers in `InitSpiBus` [here](https://github.com/nanoframework/nf-interpreter/blob/develop/targets/FreeRTOS_ESP32/ESP32_WROOM_32/nanoCLR/Windows.Devices.Spi/win_dev_spi_native_Windows_Devices_Spi_SpiDevice.cpp#L104):

```
    gpio_num_t mosiPin  = (gpio_num_t)Esp32_GetMappedDevicePins( DEV_TYPE_SPI, busIndex, 0);
    gpio_num_t misoPin  = (gpio_num_t)Esp32_GetMappedDevicePins( DEV_TYPE_SPI, busIndex, 1);
    gpio_num_t clockPin = (gpio_num_t)Esp32_GetMappedDevicePins( DEV_TYPE_SPI, busIndex, 2);
```

In fact, it appears there's [only one SPI interface](https://github.com/nanoframework/nf-interpreter/blob/develop/targets/FreeRTOS_ESP32/ESP32_WROOM_32/common/Esp32_DeviceMapping.cpp#L16) currently available in [Esp32_DeviceMapping.cpp](https://github.com/nanoframework/nf-interpreter/blob/develop/targets/FreeRTOS_ESP32/ESP32_WROOM_32/common/Esp32_DeviceMapping.cpp):

```
//  SPI 
//  2 devices
//  Map pins  mosi, miso, clock
// 
int8_t Esp32_SPI_DevicePinMap[2][3] = 
{
    {GPIO_NUM_23, GPIO_NUM_25, GPIO_NUM_19},    // SPI1 - Wrover SPI display pins
    {-1, -1, -1}                                // SPI2 - no pins assigned
};
```

Other SPI pins can be passed from C#, so why not allow all of them?

josesimoes, 2020-01-30:
cc @AdrianSoundy

AdrianSoundy, 2020-02-01:
Currently you can set the SPI pins used by the SPIx device by using the nanoFramework.Hardware.Esp32 namespace. This allows the pin mapping of all available devices on the ESP32 to be changed not just SPI.  This is a general way for all devices so no need to change the interface for the SPI uwp managed classes.

So for example to change the SPI2 pins which by default are not defined due to other device conflicts.
```
Configuration.SetPinFunction(17, DeviceFunction.SPI2_CLOCK);
Configuration.SetPinFunction(5, DeviceFunction.SPI2_MISO);
Configuration.SetPinFunction(4, DeviceFunction.SPI2_MOSI);
```

Later on i am thinking of persisting this mapping so that its automatically loaded on boot. Maybe loaded when the firmware is flashed so it is device/board specific.

gojimmypi, 2020-02-01:
@AdrianSoundy that sounds like a reasonable idea - although I wonder if the most graceful solution is to assign one pin in the `SpiConnectionSettings` object, and then set others with `Configuration.SetPinFunction`?

Even with that reassignment capability, I'm still not sure how to assign the  `RES` (reset) and `DC` (data command) pins. They don't seem to have options for the `enum DeviceFunction`.

For example, for HSPI, how to specify pins 26 `oled_dc` and 27 `oled_resn` here:

```    
public class OLED_SSD1331_HSPI
    {
        // see sample ST7735 https://github.com/nanoframework/nf-Community-Contributions/tree/master/drivers/TFT-ST7735-display
        //
        // HSPI see https://github.com/espressif/arduino-esp32/blob/7d7824701fe5e22f08555d3e1ce3180a922b2151/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino#L37
        // initialise hspi with default pins SCLK = 14, MISO = 12, MOSI = 13, SS = 15
        // working WROOM-32 using HSPI 
        byte oled_csn = 15; // white aka cs - chip select
        byte oled_dc = 26; // orange aka ds aka a0 -  SPI data or command selector pin
        byte oled_resn = 27; // gray aka rst - reset
        byte oled_mosi = 13; // purple aka mosi - data
        byte oled_clk = 14; // blue aka sclk - clock
        private  SpiDevice _spi;
        
        public void init()
        {
            var spiSettings = new SpiConnectionSettings(oled_csn);
            spiSettings.DataBitLength = 8;
            spiSettings.ClockFrequency = 10000000;
            spiSettings.Mode = SpiMode.Mode0;

            _spi = SpiDevice.FromId("SPI2", spiSettings);

            Configuration.SetPinFunction(oled_clk, DeviceFunction.SPI2_CLOCK);
            Configuration.SetPinFunction(oled_mosi, DeviceFunction.SPI2_MOSI);
        }

    }
```

AdrianSoundy, 2020-02-02:
@gojimmypi  The DC and Reset pins are not part of the SPI bus. They are specific to the displays.
You have to create separate gpio pins for those functions.

gojimmypi, 2020-02-02:
@AdrianSoundy yes, good point. I agree that DC and Reset probably don't belong in the SPI declaration.

Instead of the `Configuration.SetPinFunction`, what do you think of something like this:

```
        public void init()
        {
            var spiSettings = new SpiConnectionSettings(oled_csn, oled_clk, oled_mosi, oled_miso);
            spiSettings.DataBitLength = 8;
            spiSettings.ClockFrequency = 10000000;
            spiSettings.Mode = SpiMode.Mode0;

            _spi = SpiDevice.FromId("SPI2", spiSettings);
        }
```
and/or
```
        public void init()
        {
            var spiSettings = new SpiConnectionSettings(oled_csn);
            spiSettings.DataBitLength = 8;
            spiSettings.ClockFrequency = 10000000;
            spiSettings.Mode = SpiMode.Mode0;
            spiSettings.MosiPin = oled_mosi;
            spiSettings.MisoPin  = oled_miso
            spiSettings.ClockPin = oled_clk;

            _spi = SpiDevice.FromId("SPI2", spiSettings);
        }
```

gojimmypi, 2020-02-02:
some additional info that I posted in the [discord UI channel](https://discordapp.com/channels/478725473862549535/532862286797471744)... I was able to get the SPI somewhat working, but I need to manually control D/C line - as there appears to be no way to tell the underlying system which pin to use:

```
        private GpioPin _chipSelectLine;
        private GpioPin _DC;

        public void init()
        {
            var spiSettings = new SpiConnectionSettings(oled_csn);
            spiSettings.DataBitLength = 8;
            spiSettings.ClockFrequency = 160000; // 10000000
            spiSettings.Mode = SpiMode.Mode0;

            _chipSelectLine = GpioController.GetDefault().OpenPin(oled_csn) ;
            _chipSelectLine.SetDriveMode(GpioPinDriveMode.Output);

            _DC = GpioController.GetDefault().OpenPin(oled_dc);
            _DC.SetDriveMode(GpioPinDriveMode.Output);

            _spi = SpiDevice.FromId("SPI2", spiSettings);

            Configuration.SetPinFunction(oled_clk, DeviceFunction.SPI2_CLOCK);
            Configuration.SetPinFunction(oled_mosi, DeviceFunction.SPI2_MOSI);
        }

```
and here's where I manually trigger DC line to write a command
```
        private void WriteCommand(byte command)
        {
            _buffer1[0] = command;

            // the time between the next two stamements is a whopping 0.3ms
            // _DC.Write(GpioPinValue.Low);
            // _DC.Write(GpioPinValue.High);

            // here we manually bring DS low
            _DC.Write(GpioPinValue.Low);
            _spi.Write(_buffer1); // all 8 bits in about 54uS
            _DC.Write(GpioPinValue.High);
        }
```
note how long the D/C signal takes in comparison to the native transfer: 

[image]

josesimoes, 2020-02-21:
Guess this is not an issue anymore.
Closing.

closed 2020-02-21 by josesimoes