nanoframework/Home #561: X509 Certificate ...

valoni, 2020-01-30:
Please read the following information before posting the issue.

# **Before posting the issue**
working sample of MQTT / Mosquitto with Certificate - do not work

https://github.com/nanoframework/Samples/blob/master/samples/MQTT/TestMqtt/Mosquito.TemperatureGauge/Program.cs

because one line 27 when 
try to use converting string to byte[] cause crash of parsing certificate 

       static byte[] GetCertificate() fail on parsing ..
  
failure cause on the line 
     X509Certificate caCert = new X509Certificate(GetCertificate());

but if used certificate as string (no byte[]) it accept but do not validate on test.mosquitto.org

josesimoes, 2020-01-30:
@valoni you seem to already figure out the fix for this, could you please submit a PR in the samples repo with the fix? It's much quicker than having someone making the changes as you already have them. 😉

josesimoes, 2020-02-06:
Resolved with nanoframework/Samples#29.

closed 2020-02-06 by josesimoes