nanoframework/Home #566: VSIX extension 1.4.0.1 (on VS2017)

valoni, 2020-02-07:
After update of VSIX extension from version 1.3.0.1 to 1.4.0.1 on Visual Studio 2017 (enterprise)
i face problem with old and new projects when i try to compile.

failure is as presented on pictures (regards meteada ...)
[image]

josesimoes, 2020-02-07:
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\nanoFramework\v1.0\NFProjectSystem.MDP.targets(204,5): error : Unable to compile output assembly file 'c:\nanoFramework\nf\isma-framework-core\TemporaryDummyLibrary\obj\Debug\TemporaryDummyLibrary.pe' - check parse command results.
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\nanoFramework\v1.0\NFProjectSystem.MDP.targets(204,5): error : Object reference not set to an instance of an object.
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\nanoFramework\v1.0\NFProjectSystem.MDP.targets(204,5): error :    at nanoFramework.Tools.MetadataProcessor.nanoAssemblyBuilder.BuildDependencyList(MetadataToken token)
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\nanoFramework\v1.0\NFProjectSystem.MDP.targets(204,5): error :    at nanoFramework.Tools.MetadataProcessor.nanoAssemblyBuilder.Minimize()
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\nanoFramework\v1.0\NFProjectSystem.MDP.targets(204,5): error :    at nanoFramework.Tools.MetaDataProcessorTask.ExecuteCompile(String fileName)
1>C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\nanoFramework\v1.0\NFProjectSystem.MDP.targets(204,5): error :    at nanoFramework.Tools.MetaDataProcessorTask.Execute()

josesimoes, 2020-02-08:
Reproduced. Issue identified. Fix may be not that obvious...

closed 2020-02-10 by josesimoes