nanoframework/Home #568: 'System.Exception' in Windows.Storage.dll Failed to mount SDCard

Dweaver309, 2020-02-15:
Please read the following information before posting the issue.

**nanoFramework area:** (C# code | Nuget packages | Community contributions | Documentation )

**VS version 16.4.5

**VS extension version 1.4.0.6

**Firmware image version Image build @ Feb 14 2020 GNU ARM GCC v5.2.0

**Device capabilities output HAL build info: 1.4.0.0, nanoFramework running ESP32 built with IDF v3.3

## The Windows.Storage.dll does not see the SD card

A clear and concise description of what the bug is.
++++ Exception System.Exception - CLR_E_VOLUME_NOT_FOUND (1) ++++
    ++++ Message: 
    ++++ Windows.Storage.Devices.SDCard::MountSpiNative [IP: 0000] ++++
    ++++ Windows.Storage.Devices.SDCard::MountSpi [IP: 000c] ++++
    ++++ nfStorage.nfStorage::.ctor [IP: 0036] ++++
    ++++ nfStorage.Program::Main [IP: 002e] ++++
Exception thrown: 'System.Exception' in Windows.Storage.dll
Failed to mount SDCard 
Exception was thrown: System.Exception

## Detailed repro steps so we can see the same problem

1. Use this software: https://github.com/Dweaver309/nanoFrameworkStorage

2. Start the program and the error above is displayed

Adrian Soundly has tested with his SD card and got the same error

...

networkfusion, 2020-02-15:
We recently upgraded to fatfs R0.14. Have you updated ff.conf?

networkfusion, 2020-02-15:
But that should not have affected esp, so perhaps it is another problem...

Dweaver309, 2020-02-15:
Thanks for the reply I hope you find the issue.

Dweaver309, 2020-02-16:
Adrian Soundly has tested with a different SD card and the problem was solved. I assume my issue most be with the hardware.

Dweaver309, 2020-03-27:
I am still having this issue I think @AdrianSoundy  used SDCard.MountMMC(false) in his test above.
This is a link to my test code: https://github.com/Dweaver309/SPI_SDCard_Fails

piwi1263, 2020-03-27:
@Dweaver309 Thanks for the info but (assuming you have done this on ESP32) I'm trying to get this working on STM32. The SD Card using 4Bit SDIO seems to be working but I don't even get the insert event to fire in c# which might be caused by not being able to mount the card and therefore not raising a fw event and no managed event either. I'm still investigating, testing with the F769 Ref board to have at least a working setup so I can rule out the system as much as can be. Maybe @josesimoes or @networkfusion can through in some ideas how to move forward and get this solved.

Dweaver309, 2020-03-27:
Thanks Peter

Dweaver309, 2020-03-29:
I was able to get the SD card working using SPI and ESP32 Dev-C today. I was using a bread board that must have shortened out or not enough voltage? I connected the SD shown here https://github.com/Dweaver309/nanoFrameworkStorage/blob/master/ESP32SDCard.png
directly to the board with new bread board wires.

closed 2020-03-29 by Dweaver309