nanoframework/Home #570: ValueChanged event triggering stops on ESP32

kmrzyglod, 2020-02-20:
## Details about Problem

**Target:** ESP32

**Firmware image version:** 1.4.0-preview.110

## Description

I have some issues with ValueChanged event handling on ESP32 especially when DebounceTimeout is set. During debugging example code below I see that sometimes after a few triggers of ValueChanged this event stops triggering. Only reinitializing the pin is helping after that. 
If time period between pin state changes is quite long everything can work properly for long time but I noticed that when I start changing pin state (e.g. by manually connecting and disconnecting pin from GND) with some higher frequency like 1-2 times per second this issue can occur after just a few triggers. If DebounceTimeout isn't set this issue occurs less times but still. This can be problematic becuse I have not certainty that some time after launch my program this issue will not break e.g. detecting reed switches state changes or counting precipitation from rain gauge (because I tried to use ValueChanged event in rain gauge driver for my weather station project). 
I am using external pull-up 47k resistor because with internal pull-up it seems that issue is occuring more often.

## Detailed repro steps 

1. Start debugging this simple code with external pull-up resistor attached to GPIO23:
```var gpioController = GpioController.GetDefault();
var pin = gpioController.OpenPin(23);
pin.SetDriveMode(GpioPinDriveMode.Input);
pin.DebounceTimeout = TimeSpan.FromMilliseconds(50);
pin.ValueChanged += (_, eventArgs) =>
{
    Console.WriteLine($"State changed: {eventArgs.Edge}");
};
```
2. Try to manually connecting and disconnecting pin from GND 1-2 times per second
3. After a few triggers ValueChanged event stops triggering

josesimoes, 2020-04-13:
@kmrzyglod can you give it another try?
This has probably been corrected with nanoframework/nf-interpreter#1598.

kmrzyglod, 2020-04-14:
@josesimoes I've tried to reproduce this issue using newest firmware and nugets and I can't so probably bug was fixed and this issue can be closed.

josesimoes, 2020-04-15:
Fixed with nanoframework/nf-interpreter#1598

closed 2020-04-15 by josesimoes