nanoframework/Home #571: I2S library

Hasaken, 2020-02-20:
**nanoFramework area:** 
Hello.

I couldn't find I2S protocol support in nanoFramework. Do you plan to add it in close future?

josesimoes, 2020-02-20:
That is certenaly possible. 
There is currently no date for this.

Questions:
- What platform are you using?
- Would you ne willing to help?

Hasaken, 2020-02-20:
Thanks for the quick response.

I use ESP32.
I can help by testing.

josesimoes, 2020-07-02:
Hasn't picked up much interest from the community.
Adding to TODO list.
Closing issue.

josesimoes, 2021-09-23:
High level task list to make this real:

- [ ] Add I2S GPIO assignment to nanoFramework.Hardware.Esp32.
- [ ] Update native part to support the above changes:
   targets\FreeRTOS_ESP32\ESP32_WROOM_32\common\Esp32_DeviceMapping.cpp
   targets\FreeRTOS_ESP32\ESP32_WROOM_32\Include\Esp32_DeviceMapping.h
- [ ] Add new class library for I2S
   Agree on name, create new repo, agree on API
- [ ] Add native part for this new class library
   CMake, build options, update templates, add common library, add platform implementation

Ellerbach, 2021-09-25:
See https://github.com/nanoframework/nf-interpreter/pull/2073

See https://github.com/nanoframework/nanoFramework.Hardware.Esp32/pull/145

Ellerbach, 2022-03-03:
Still valid

Ellerbach, 2022-05-02:
Still on!

AdrianSoundy, 2022-08-20:
@Ellerbach @Schmendric 
Is this still being worked on ?
Otherwise i can look at competing it.

Ellerbach, 2022-08-22:
> Otherwise i can look at completing it.

Feel free to move this forward. There is a branch on nf-interpreter where the initial native work is there. Need to be updated with the new build system.

The manage code exist as well.

All *should* almost work properly. So *should* be quite straight forward! Ping me directly if you need help!

AdrianSoundy, 2022-08-23:
> Feel free to move this forward. There is a branch on nf-interpreter where the initial native work is there. Need to be updated with the new build system.
>
> The manage code exist as well.
>
> All [...]

Yes i saw how much was done. I'll  take it on and see if i can at least finish initial version.

Ellerbach, 2022-08-23:
> Yes i saw how much was done. I'll take it on and see if i can at least finish initial version.

I almost did everything, just missing a test. I was blocked before the last major IDF migration because there was no prebuild lib. And since then, no time to take it over. But manage code, even the sample to test, everything should be there. Take extra care of the modified files (like the hardware ESP32 ones, I remember I did few changes there as well on top of the I2S files per se). And please do not hesitate to reach out!

closed 2022-10-31 by josesimoes