nanoframework/Home #572: VSIX do not detect STM32 boards

valoni, 2020-02-25:
After update-in VSIX 1.7.0.6 and installed firmware for NUCLEO F411 version 1.1.328 .0
when tried to detect boards showing 

VSIX do not detect board a

message : 
**2020-02-25 22:03:50Z [NanoDevices: checking device  \\?\USB#VID_0483&PID_374B&MI_02#6&3381915F&1&0002#{86e0d1e0-8089-11d0-9ce4-08003e301f73} with invalid DeviceBase]**

it also confirmed with official board too Netduino3_Wifi firmware 1.4.0.preview.115

[image]

josesimoes, 2020-02-26:
After several tests I've come to the following conclusion: this seems to happen with STM boards that use the STLink USB connection **and** the firmware that includes Mass Storage.

If you install the firmware **without** Mass Storage option the board detection works as expected.

[image]
✔️  Funcional

[image]

🔴 NOT funcional

Further investigation is required...

josesimoes, 2020-02-26:
Recommendation to update firmware on boards with STM32 STLink to v2.36.26.

https://www.st.com/content/st_com/en/products/development-tools/software-development-tools/stm32-software-development-tools/stm32-programmers/stsw-link007.html

josesimoes, 2020-02-26:
NETDUINO3_WIFI is not enumerating USB device properly. It's an issue, of course, but should be unrelated.

josesimoes, 2020-02-26:
Should be fixed with VS extension v1.7.0.9.

closed 2020-02-26 by josesimoes