nanoframework/Home #574: Storage sample on ESP32

phil31, 2020-02-27:
**nanoFramework area : sample "storage"

System Information
HAL build info: 1.4.0.0, nanoFramework running ESP32 built with IDF v3.3
Target: ESP32_WROOM_32

## Description
trying the sample with the **internal flash memory of the module** - the 3 SDcard configurations lines in scenario 0, are commented - NO SDcard mounted
so it jump directly after the Thread.Sleep(3000); to scenario 1 

here it is my ouput window 👍 
== Scenario0_mountDevice ==
== GetRemovableDevices ==
**== Scenario1_CreateAFolderInStorage ==**
    ++++ Exception System.ArgumentException - 0xfd000000 (1) ++++
    ++++ Message: 
    ++++ Windows.Storage.StorageFolder::CreateFolderNative [IP: 0000] ++++
    ++++ Windows.Storage.StorageFolder::CreateFolder [IP: 0006] ++++
    ++++ FileAccess.Scenario1_CreateAFolderInStorage::Execute [IP: 0011] ++++
    ++++ FileAccess.Program::Main [IP: 0048] ++++
Exception levée : 'System.ArgumentException' dans Windows.Storage.dll
ERROR: can't create the folder as it already exists.
**== Scenario2_CreateAFileInStorage ==**
    ++++ Exception System.Exception - CLR_E_FILE_NOT_FOUND (1) ++++
    ++++ Message: 
    ++++ Windows.Storage.StorageFolder::CreateFileNative [IP: 0000] ++++
    ++++ Windows.Storage.StorageFolder::CreateFile [IP: 0006] ++++
    ++++ FileAccess.Scenario2_CreateAFileInStorage::Execute [IP: 0011] ++++
    ++++ FileAccess.Program::Main [IP: 004e] ++++
Exception levée : 'System.Exception' dans Windows.Storage.dll
Error: Unable to create file: Exception was thrown: System.Exception
**== Scenario3_WriteAndReadTextInAFile ==**
    ++++ Exception System.Exception - CLR_E_FILE_NOT_FOUND (1) ++++
    ++++ Message: 
    ++++ Windows.Storage.StorageFolder::CreateFileNative [IP: 0000] ++++
    ++++ Windows.Storage.StorageFolder::CreateFile [IP: 0006] ++++
    ++++ FileAccess.Scenario3_WriteAndReadTextInAFile::Execute [IP: 0012] ++++
    ++++ FileAccess.Program::Main [IP: 0054] ++++
Exception levée : 'System.Exception' dans Windows.Storage.dll

scenario 1 pretend that the directory is already existing ..i'm sure not. i change the name of the folder with a new one (wich i'm certain that can't exist); same result

scenario 2 : exception .. 

scenario 3 : it stop the debbug mode with the exception

josesimoes, 2020-02-27:
Looks like you don't have the latest fw installed on your device. Care to update and try again?

Also please post here the full output of the device capabilities.

phil31, 2020-02-28:
using EspFirmwareFlasher.exe, i update the module to 1.4.0-preview.118 and re run the sample.

exactly the same, same exception :(

----------------------------------------------------------------------------------------------------------
System Information
HAL build info: 1.4.0.0, nanoFramework running ESP32 built with IDF v3.3.1
Target: ESP32_WROOM_32
Platform: ESP32

Image build @ Feb 27 2020 GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Solution Build Info: 1.4.0.0, nanoFramework running ESP32 built with IDF v3.3.1

AppDomains:

Assemblies:
  FileAccess, 1.0.0.0
  nanoFramework.Hardware.Esp32, 1.2.0.0
  Windows.Storage, 1.2.0.0
  System.Net, 1.4.1.0
  Windows.Devices.I2c, 1.3.1.0
  mscorlib, 1.6.0.0
  nanoFramework.System.Collections, 1.0.0.0
  Windows.Devices.Spi, 1.3.0.0
  Windows.Storage.Streams, 1.8.0.0
  Windows.Devices.Wifi, 1.2.0.0
  Windows.Devices.Gpio, 1.4.1.0
  nanoFramework.Runtime.Events, 1.4.1.0
  nanoFramework.System.Text, 1.0.0.0
  System.Net.Http, 1.2.2.0
  nanoFramework.Runtime.Native, 1.2.0.0

Native Assemblies:
  mscorlib v100.4.6.0, checksum 0x3E5F09B0
  nanoFramework.Runtime.Native v100.0.6.0, checksum 0x035EC7E8
  nanoFramework.Hardware.Esp32 v100.0.7.0, checksum 0xC16385A2
  nanoFramework.Devices.OneWire v100.0.3.3, checksum 0xC7BEAEBC
  nanoFramework.Networking.Sntp v100.0.4.3, checksum 0xB617FDFB
  nanoFramework.ResourceManager v100.0.0.0, checksum 0xCDF9E76C
  nanoFramework.System.Collections v100.0.0.0, checksum 0xC7167094
  nanoFramework.System.Text v100.0.0.0, checksum 0x482137FC
  nanoFramework.Runtime.Events v100.0.7.0, checksum 0x7B4AC2F2
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.3, checksum 0xC5A2A195
  System.Net v100.1.3.0, checksum 0x0B718489
  Windows.Devices.Adc v100.1.3.2, checksum 0xDEDF1136
  Windows.Devices.Gpio v100.1.2.0, checksum 0x2432E7EB
  Windows.Devices.I2c v100.2.0.1, checksum 0x8D1C8F95
  Windows.Devices.Pwm v100.1.3.2, checksum 0x84D8B10C
  Windows.Devices.SerialCommunication v100.1.1.0, checksum 0x8E46E2E2
  Windows.Devices.Spi v100.1.4.0, checksum 0xEA555AF2
  Windows.Devices.Wifi v100.0.6.0, checksum 0x5AC050C9
  Windows.Storage v100.0.1.0, checksum 0x8D8C8026

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffceb24  0x00010000
  FLASH 0x08000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x180000     nanoCLR
      1    0x00190000       1      0x1B0000     Deployment
      2    0x00340000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x00340000    0x040000 (256kB)    Configuration
  0x00010000    0x180000 (1536kB)   nanoCLR
  0x00190000    0x1B0000 (1728kB)   Deployment

Deployment Map
Empty

-----------------------------------------------------------------------------------------------------------

networkfusion, 2020-03-03:
@phil31 given your comment on discord, is this a non issue?

phil31, 2020-03-03:
sorry for this ticket, i do some mistake

issue can be closed

closed 2020-03-03 by phil31