nanoframework/Home #575: Enum in class in library not working with VS Extension >= 1.3.0.0

MateuszKlatecki, 2020-02-28:
## Details about Problem

**nanoFramework area:**  Visual Studio extension 

**VS version:** 2017

**VS extension version:** tested with 1.7.0.9

**Target:** 

**Firmware image version:**

**Device capabilities output:**

## Description

If I create library that contains public class and inside this class I create public enum and then in my app add reference to this lib and try to use that enum everything is compile ok but program didn't run. With extension 1.2.5.38 program works.

Trying to debug the problem, it turns out that the program does not see the Enum type (CLR_RT_Assembly::Resolve_TypeRef() function)

## Expected behaviour

Program should work when compiled with new extension

## Additional context

Attached solution causing problem
[NFFail.zip](https://github.com/nanoframework/Home/files/4265882/NFFail.zip)

josesimoes, 2020-02-28:
There was a fix with enums in classes in the last version of MDP.
Please update VS extension to 1.7.0.11 and verify if this is still an issue.

MateuszKlatecki, 2020-02-28:
in 1.7.0.11 fail to deploy:
```
2020-02-28 11:22:31Z [Check and start debug engine on nanoDevice.]
2020-02-28 11:22:31Z [Connect successful.]
2020-02-28 11:22:31Z [Erase deployment area successful.]
2020-02-28 11:22:31Z [Device status verified as being in initialized state. Requesting to resume execution.]
2020-02-28 11:22:31Z [Device has completed initialization.]
2020-02-28 11:22:31Z [Computing deployment blob.]
2020-02-28 11:22:33Z [Found assemblies mismatches when checking for deployment pre-check.]
2020-02-28 11:22:33Z [Exception occurred during deployment.
 Deploy failed.
***************************************
Couldn't find a valid native assembly required by NFClassLibrary1 v1.0.0.0, checksum 0x0AAD7E0C.
This project is referencing NFClassLibrary1 NuGet package requiring native v.
The connected target does not have support for NFClassLibrary1.
Please check: 
  1) if the target is running the most updated image.
  2) if the target image was built to include support for all referenced assemblies.

If you are using PREVIEW versions of the NuGet packages, make sure you have our Azure Pipelines feed setup.
See here for details: https://nanoframework.net/2018/05/16/setup-visual-studio-to-access-preview-versions-feed

*************************************** 
  
    at nanoFramework.Tools.VisualStudio.Extension.DeployProvider.<DeployAsync>d__17.MoveNext()]

```

but nFClassLibrary1 isn't native :|

josesimoes, 2020-02-28:
This has nothing to do with the enum, rather with the deployment check that's performed for native support of a library.

Need to figure out a way of dealing with these situations.

MateuszKlatecki, 2020-02-28:
In 1.7.0.9 deploy was successfull but program not run. After upgrade extension can't deploy program. I think update brings new bug but i can't confirm that bug reported initially is fixed due the new bug.

closed 2020-02-28 by josesimoes