nanoframework/Home #581: Using SDCard

MHSchmidt, 2020-03-05:
VS Extension 1.18.0.1
Image 1.4.0-preview.122
When I deploy the application I get the following error:

1>Getting things ready to deploy assemblies to nanoFramework device: ST_STM32F769I_DISCOVERY @ COM11.
1>nanoDevice is initialized
1>Deploy failed.
1>***************************************
1>Couldn't find a valid native assembly required by Windows.Storage.Streams v1.8.1.0, checksum 0x5C0DB88E.
1>This project is referencing Windows.Storage.Streams NuGet package requiring native v.
1>The connected target does not have support for Windows.Storage.Streams.

josesimoes, 2020-03-06:
Turns out that the Azure Pipeline was pulling in an outdated version of MDP.
Task now updated to 1.201. Builds are now using the latest version.

This will go away as soon as the next NuGet packages are published.

closed 2020-03-06 by josesimoes

josesimoes, 2020-03-06:
The new NuGets are now available and this should be usable again.