nanoframework/Home #583: Raspberry pi

furesoft, 2020-03-12:
Can i Use NanoFramework on Raspberry pi?

networkfusion, 2020-03-12:
Not currently,  although it is technically possible. Although since the raspberry pi can run the full .net core framework, I am not sure what you would achieve by doing so... This type of question would be better answered on our discord channel if further comment is required.

closed 2020-03-12 by josesimoes

asheesh1996, 2022-04-20:
Can I Use NanoFramework on Raspberry pi and configure wifi using System.Devices.Wifi package as well as any chance for BLE?

Ellerbach, 2022-04-20:
> Can I Use NanoFramework on Raspberry pi and configure wifi using System.Devices.Wifi package as well as any chance for BLE?

No, you cannot use .NET nanoFramework on a Raspberry Pi. But you can use .NET IoT, see https://github.com/dotnet/iot/. API are aligned, so you can almost use the same code on both platforms.

asheesh1996, 2022-05-11:
> No, you cannot use .NET nanoFramework on a Raspberry Pi. But you can use .NET IoT, see https://github.com/dotnet/iot/. API are aligned, so you can almost use the same code on both platforms.
As I mentioned in my question, I want to use [System.Devices.Wifi](https://www.nuget.org/packages/nanoFramework.System.Device.Wifi/) package and [BLE package](https://github.com/nanoframework/nanoFramework.Device.Bluetooth) on raspberry pi and there are not much libraries for these functionalities.

josesimoes, 2022-05-11:
> As I mentioned in my question, I want to use [System.Devices.Wifi](https://www.nuget.org/packages/nanoFramework.System.Device.Wifi/) package and [BLE [...]

These libraries are meant to be used in .NET nanoFramework projects. To my knowledge there are no equivalents for .NET that you can use in the RPII. 

Suggest that you ask on .NET IoT repo or their [Discord](https://discord.gg/YD58GfFS) server.