nanoframework/Home #584: Crash at start of Main() function after integrating some nuget DLLs

skyflashde, 2020-03-13:
Everything was running fine with the latest DLLs, then I added the networking sample to the code to check it out.

Now my program crashes before any breakpoint I can set, before my Main() gets called.

++++ Exception System.ArgumentNullException - CLR_E_ARGUMENT_NULL (2) ++++
    ++++ Message: 
    ++++ System.Threading.Monitor::Enter [IP: 0000] ++++
    ++++ System.Net.WebRequest::RegisterPrefix [IP: 0018] ++++
    ++++ System.Net.HttpWebRequest::.cctor [IP: 0028] ++++
An unhandled exception of type 'System.ArgumentNullException' occurred in mscorlib.dll

    ++++ Exception System.ArgumentNullException - CLR_E_ARGUMENT_NULL (2) ++++
    ++++ Message: 
    ++++ System.Threading.Monitor::Exit [IP: 0000] ++++
    ++++ System.Net.WebRequest::RegisterPrefix [IP: 0052] ++++
    ++++ System.Net.HttpWebRequest::.cctor [IP: 0028] ++++
An unhandled exception of type 'System.ArgumentNullException' occurred in mscorlib.dll

I am not using any HttpWebRequest in my program, so it must be called by some DLL I don't currently use but have referenced.

[image]

josesimoes, 2020-03-14:
This is happening because the variable used in the lock hasn't been initialized.
Possible cause has already been reported in #530.

skigrinder, 2020-03-15:
Having the same issue.    
Interestingly, when not connected for debug, my ESP32 board boots normally and handles http requests as per my code.  
When connected for debug, I can 'continue execution' at each of the two errors.
Then my code comes up and runs fine.

phil31, 2020-03-29:
hi, any update about this issue ?
i have it too !
so i decide to create a new simple "hello world"solution and add some references.
i 've got his issue and don't know which variable Joses speak about ..?

[image]

please any tips/solution ??

skigrinder, 2020-03-29:
> image

Hi Phil,
I don't think anything has happened with this and I'm currently pounding 
on the JSON library.
It looks like a tough problem that's probably way beyond what I could 
handle.
It happens twice in my solution when I start Debugging.
However, the screen that pops up offers a link to 'continue execution'.
If I hit this, my code comes up and I'm able to keep working.
It only happens in my solutions that where I'm using:
     using System.Net;
     using System.Net.NetworkInformation;
     using Windows.Devices.WiFi;

I suspect it also happens with System.Net.Sockets.
Have you tried commenting this out?
Are you running in Debug mode?
Do you get the screens I mentioned above?
Thanks,
    Jim - skigrinder

phil31, 2020-03-30:
Hi Jim,

I successively removed different references.
then now, no more error, "hello world" is running !

yes i use the debug mode and i can "continue" when i get the previous error/exception .. but i'm not confident in that case !..?

i just add back nanoframework.system.net and system.net.http then the issue arrive again !..
(before that, i jsut add .json : no problem)

I'm already having trouble writing C # correctly, so finding a solution to this problem is way beyond my knowledge!

[image]

josesimoes, 2020-03-30:
The variable that is not being initialized as it should is this one:
https://github.com/nanoframework/lib-nanoFramework.System.Net.Http/blob/ab5b5c81ca5afd3d544aa392cc37b073f635ba5c/source/nanoFramework.System.Net.Http/Http/System.Net.WebRequest.cs#L28

The most likely reason for it is that something wrong is happening in the start execution workflow. Before the app starts to execute one of the preparing steps is that all static constructors (on all classes) are executed so that the static vars (like the one above) are initialized. 
When that fails leads to what you're seeing: when that var is accessed it is (wrongly) null and can't be used.

josesimoes, 2020-03-30:
And, yes, nothing has happen on this so far because I had no time to look into this and no one else has step forward to do it either.

phil31, 2020-03-30:
thanks Jose for your explanation.
what are your recomandations about the issue ? can we forget it for now ?

josesimoes, 2020-03-30:
@phil31 it has to be fixed. No specific recommendation... The impact on this depends on "how quick" and/or the order which those constructors are executed. If they are executed before the class is used it should work as expected. If not, the most likely result, is that exceptions will be thrown and the app may fail. All depending if it's using the affected classes or not.

closed 2020-04-29 by josesimoes

skigrinder, 2020-04-30:
> Closed #584 <https://github.com/nanoframework/Home/issues/584> via
> nanoframework/nf-interpreter#1619

Great stuff Jose' !!
All of my code that hit this problem is working well now.
Networkfusion found a few of my boneheaded issues in the JSON code and 
we're working through them.
Hopefully, the author will give us permission to use the library.
Thanks,
    Jim

josesimoes, 2020-04-30:
Glad that's working for you now!