nanoframework/Home #585: M2MqTT Not Working

ivan0414, 2020-03-19:
Hi, I have an issue with nanoframework.M2Mqtt library version 4.4.0-preview.10 (it is latest on nuget)

My device also updated to latest firmware version - 1.4.0 Preview 141 and VS Extension updated

Device Capabilities:

System Information
HAL build info: nanoFramework running @ ESP32
  Target:   ESP32_WROOM_32
  Platform: ESP32

Firmware build Info:
  Date:     Mar 15 2020
  Type:     MinSizeRel build with IDF v3.3.1
  Version:  1.4.0.0
  Compiler: GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

AppDomains:

Assemblies:
  NFTestNUGET, 1.0.0.0
  mscorlib, 1.7.0.0
  nanoFramework.Runtime.Events, 1.4.2.0
  nanoFramework.M2Mqtt, 4.3.1.2
  nanoFramework.System.Text, 1.0.0.0
  nanoFramework.Runtime.Native, 1.2.0.0
  nanoFramework.System.Collections, 1.0.0.0
  nanoFramework.Hardware.Esp32, 1.2.1.0
  Windows.Devices.Gpio, 1.4.1.0
  System.Net, 1.4.1.0

Native Assemblies:
  mscorlib v100.4.7.0, checksum 0x42B9EEF6
  nanoFramework.Runtime.Native v100.0.6.1, checksum 0x6792B404
  nanoFramework.Hardware.Esp32 v100.0.7.1, checksum 0x1B75B894
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xC7BEAEBC
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xB617FDFB
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xE31F7E61
  nanoFramework.System.Collections v100.0.0.1, checksum 0x3D26C4A9
  nanoFramework.System.Text v100.0.0.1, checksum 0x482137FC
  nanoFramework.Runtime.Events v100.0.7.1, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.4, checksum 0xC5A2A195
  System.Net v100.1.3.1, checksum 0x1118F266
  Windows.Devices.Adc v100.1.3.3, checksum 0xDEDF1136
  Windows.Devices.Gpio v100.1.2.1, checksum 0x25180D57
  Windows.Devices.I2c v100.2.0.2, checksum 0x754E3590
  Windows.Devices.Pwm v100.1.3.3, checksum 0x84D8B10C
  Windows.Devices.SerialCommunication v100.1.1.1, checksum 0x82260711
  Windows.Devices.Spi v100.1.4.1, checksum 0xEA555AF2
  Windows.Devices.Wifi v100.0.6.1, checksum 0xDF2FD922
  Windows.Storage v100.0.2.0, checksum 0xE6682807

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffcea6c  0x00010000
  FLASH 0x08000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x180000     nanoCLR
      1    0x00190000       1      0x1B0000     Deployment
      2    0x00340000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x00340000    0x040000 (256kB)    Configuration
  0x00010000    0x180000 (1536kB)   nanoCLR
  0x00190000    0x1B0000 (1728kB)   Deployment

Deployment Map
Empty

System Information
HAL build info: nanoFramework running @ ESP32
  Target:   ESP32_WROOM_32
  Platform: ESP32

Firmware build Info:
  Date:     Mar 15 2020
  Type:     MinSizeRel build with IDF v3.3.1
  Version:  1.4.0.0
  Compiler: GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

AppDomains:

Assemblies:
  NFTestNUGET, 1.0.0.0
  mscorlib, 1.7.0.0
  nanoFramework.M2Mqtt, 4.4.0.0
  nanoFramework.Runtime.Events, 1.4.2.0
  nanoFramework.System.Text, 1.0.0.0
  nanoFramework.Runtime.Native, 1.2.0.0
  nanoFramework.System.Collections, 1.0.0.0
  nanoFramework.Hardware.Esp32, 1.2.1.0
  Windows.Devices.Gpio, 1.4.1.0
  System.Net, 1.4.1.0

Native Assemblies:
  mscorlib v100.4.7.0, checksum 0x42B9EEF6
  nanoFramework.Runtime.Native v100.0.6.1, checksum 0x6792B404
  nanoFramework.Hardware.Esp32 v100.0.7.1, checksum 0x1B75B894
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xC7BEAEBC
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xB617FDFB
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xE31F7E61
  nanoFramework.System.Collections v100.0.0.1, checksum 0x3D26C4A9
  nanoFramework.System.Text v100.0.0.1, checksum 0x482137FC
  nanoFramework.Runtime.Events v100.0.7.1, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.4, checksum 0xC5A2A195
  System.Net v100.1.3.1, checksum 0x1118F266
  Windows.Devices.Adc v100.1.3.3, checksum 0xDEDF1136
  Windows.Devices.Gpio v100.1.2.1, checksum 0x25180D57
  Windows.Devices.I2c v100.2.0.2, checksum 0x754E3590
  Windows.Devices.Pwm v100.1.3.3, checksum 0x84D8B10C
  Windows.Devices.SerialCommunication v100.1.1.1, checksum 0x82260711
  Windows.Devices.Spi v100.1.4.1, checksum 0xEA555AF2
  Windows.Devices.Wifi v100.0.6.1, checksum 0xDF2FD922
  Windows.Storage v100.0.2.0, checksum 0xE6682807

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffcea6c  0x00010000
  FLASH 0x08000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x180000     nanoCLR
      1    0x00190000       1      0x1B0000     Deployment
      2    0x00340000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x00340000    0x040000 (256kB)    Configuration
  0x00010000    0x180000 (1536kB)   nanoCLR
  0x00190000    0x1B0000 (1728kB)   Deployment

Deployment Map
Empty

The error is:

** Error: Device stopped after type resolution failure.                     **
The program '[1] nanoFramework application: Managed' has exited with code 0 (0x0).
** Check in each assembly which assemblies are referenced and their version **
******************************************************************************
The nanoDevice target and the debugger engine failed to initialize because of unspecified device errors.
The debugger engine thread has terminated unexpectedly with error 'Device stopped after type resolution failure. Can't start execution.'.

josesimoes, 2020-03-20:
This is being caused by a mismatch with the referenced class libraries.

We are currently trying to improve how the assembly resolution is handled by VS and make sure that any mismatches produce a build error so they are made perfectly clear and can be fixed by the developer.

ivan0414, 2020-03-20:
Thank you for the response. I downgraded the firmware of the esp32 to an older version and also downgraded the Core.Library to 1.6.0 and all its dependencies and now its working fine with the M2Mqtt version 4.4.0-preview.10.

networkfusion, 2020-04-19:
Tested and working with the most recent nuget releases on an stm32f769i,  is this still an issue for the ESP32?

josesimoes, 2020-04-21:
Closing this for lack of feedback. Hopefully the issue is now solved. If not can be reopened later.

closed 2020-04-21 by josesimoes