nanoframework/Home #586: Gpio interrupts have stopped working for ESP32 (and STM32?)

skigrinder, 2020-03-20:
The following code was working a few weeks ago but it no longer triggers the event handler when the device is activated:

            using Windows.Devices.Gpio;
            using nanoFramework.Hardware.Esp32;

            private static GpioPin FC51_01;
          
            FC51_01 = gpioController.OpenPin(Gpio.IO19, GpioSharingMode.SharedReadOnly);
            FC51_01.DebounceTimeout = new TimeSpan(0, 0, 0, 0, 50);
            FC51_01.SetDriveMode(GpioPinDriveMode.InputPullDown);
            FC51_01.ValueChanged += FC51_01_gpioValueChanged;

I've tested this for ESP32 with the latest nanoFramework VS extension & Nuget libraries.
Changes to the DebounceTimeout & GpioPinDriveMode did not resolve the issue.

piwi1263, 2020-03-22:
Hi @josesimoes and @skigrinder I guess I have the same issue or at least similar. After updating the Quail (STM32) board support for the microSD Click board only the remove event is fired, the insert event never does.

martin-kuhn, 2020-03-30:
Yes, my button interrupt does not work anymore either on STM32 with the latest version.

josesimoes, 2020-03-30:
I've managed to reproduce the issue. Working on it. Please be patient.

closed 2020-04-02 by josesimoes