nanoframework/Home #589: Json - DeserializeString() result can't be cast to a class object

skigrinder, 2020-03-28:
The following code (condensed) from the JsonConfigurationStore sample no longer works.
JsonSerializer.DeserializeString() throws  System.InvalidCastException - CLR_E_INVALID_CAST (1)

        public class Configuration {
            public string Setting1      { get; set; }
            public string Setting2      { get; set; }
            public string Setting3      { get; set; }
        }

        public static void Main() {
            Configuration configuration = new Configuration() {
                Setting1 = "Setting 1 value",
                Setting2 = "Setting 2 value",
                Setting3 = "Setting 3 value"
            };
            var Jstring = JsonSerializer.SerializeObject(configuration);
            Configuration config = (Configuration)JsonSerializer.DeserializeString(Jstring);
        }

Looks like the ability to read json data back into a class object is broken.

networkfusion, 2020-03-28:
Given this is not a core library for the project, this seems to be a missing feature, that is unlikely to be fixed by the core team. There is a project for .netmf that could be ported instead and the team would be willing to change to it if it was... see https://discordapp.com/channels/478725473862549535/481783110779011083/693541185284734986 and https://discordapp.com/channels/478725473862549535/481783110779011083/693548474632765451 for further details.

skigrinder, 2020-03-29:
Thanks Robin,
I've started looking at this.
Thanks,
    Jim

networkfusion, 2020-04-28:
Lets look at @skigrinder 's contribution to see how we can integrate it.

https://github.com/skigrinder/jsonnetmf

josesimoes, 2020-04-28:
I started looking at the code there and header on the code files is pretty clear:
```
// (c) Pervasive Digital LLC
// Use of this code and resulting binaries is permitted only under the
// terms of a written license.
```

We either submit a PR to the original repo and wait that's accepted or this is a no go to have this as  _the_ official nF json library...

josesimoes, 2020-04-28:
The alternative would be to port the fix implemented there to our json library... haven't looked into the the extension of the work, maybe @skigrinder input here is in order!

networkfusion, 2020-05-10:
lib-nanoFramework.Json is now available for testing

josesimoes, 2020-05-10:
V2.0 preview 😜

josesimoes, 2020-05-29:
@networkfusion isn't this working with the latest version of the new lib?

networkfusion, 2020-05-29:
It will be once the nuget is published as preview.

josesimoes, 2020-05-29:
Well that happened yesterday so we are good! 😁

https://www.nuget.org/packages/nanoFramework.Json/2.0.0-preview.18

closed 2020-05-29 by josesimoes