nanoframework/Home #590: VS Extension timing issues with external memories

martin-kuhn, 2020-03-30:
## Details about Problem

We are running nanoFramework on custom STM32F427 based MCU boards. They also contain an external flash for the deployment and external SDRAM which is used for the heap and the managed heap.

**nanoFramework area:** Visual Studio extension

**VS version:** Visual Studio 2017

**VS extension version:**

This behaviour shows up with nF extension 1.2.4.28.
I also tried to update to the newest version of the extension, but then, another related problem occurs. I can't even deploy with this version of the extension. The deployment provider reports "Object reference not set to an instance of an object". When I flash the application with ST-Link, then it starts without problems, BUT ONLY if Visual Studio is not running. If Visual Studio is running and attached to the target, the target stops after loading the second assembly and gets trapped in the WaitForInterrupts Loop.

**Target:** custom STM32F427 based MCU board

**Firmware image version:**

**Device capabilities output:**

## Description

The handling of the device in VS works fine (Deploying, ping, erase deployment area, show device caps, etc.). But we are having troubles with starting the debugger, especially with bigger applications. 
There are several problems that occur:
- sometimes the deployment provider reports "Object reference not set to an instance of an object"
- sometimes it reports "Device is reporting no assemblies deployed, can't debug."
- sometimes the debugger starts but the output is in a strange order
For me this looks like a timing problem between the target and the VS extension. The problem does not occur when the internal flash and the internal RAM is used. It's highly likely that this problem is caused by longer data acquisition times for the external memories and then we run into a timeout somewhere.

## Detailed repro steps so we can see the same problem

This issue can only be reproduced if a target with external memory (via FSMC) is used.

## Screenshot
The following screenshot shows how the output of the debugger gets messed up.

[image]

josesimoes, 2020-03-30:
I'm more inclined to the root cause being #530. Failing to run the static constructors on all classes before executing main() can lead to the behaviours that you're describing here.

patrick-haldi, 2020-04-22:
We worked out a few workarounds, and got the deployment running. The main issues that kept us from successfully deploying and debugging code were:
-Interfering debugging messages sent from CLRStartup thread
-Race condition with free’d but still used g_CLR_DBG_Debugger structure, which led to a hard fault.
-Visual Studio Extension did not wait long enough with sending messages after soft reboot of the device.
This is explained in full length in the discord message: https://discordapp.com/channels/478725473862549535/591956247767547910/702432889341739069

josesimoes, 2020-04-29:
The deployment issues you're describing are related with #603.

closed 2020-04-29 by josesimoes