nanoframework/Home #594: GpioPin.Read() not working (v1.4.1-preview.13)

opitzh, 2020-04-08:
## Details about Problem

**nanoFramework area:** Visual studio extension

**VS extension version:** 2019.1.8.9, Windows.Devices.Gpio v1.4.1-preview.13

**Target:**

**Firmware image version:**

**Device capabilities output:**

## Description
GpioPin - Read() method doesn't work with ESP32. I think due to this, also the Toggle() method is not working too. The Read() method returns all the time 0.

## Detailed repro steps so we can see the same problem
[Program.zip](https://github.com/nanoframework/Home/files/4448776/Program.zip)

## Other suggested things

## Expected behaviour

### Sample Project

## Screenshot
[image]

## Additional context

josesimoes, 2020-04-28:
@opitzh seems that you're reading from a GPIO that's configured as output. That won't work. You can only read from GPIOs configured as Input.

closed 2020-04-28 by josesimoes

josesimoes, 2020-04-29:
BTW, tested with Blinky sample and it's working as expected.