nanoframework/Home #595: i2c Not Sending text to device

Dweaver309, 2020-04-09:
## Details about Problem
The i2c library is not sending text to the device properly. I have tested with 2 different applications.

**Target:**
ESP32 Dev C

**Firmware image version:**
1.4.0-preview.155

Worked before? If so, with which nanoFramework image version:
The library worked in December 2019 don't know when it stopped working.

## Description
The attached application sends just three characters to  the display. 

[NF.SPI.ST7735.Driver.zip](https://github.com/nanoframework/Home/files/4459007/NF.SPI.ST7735.Driver.zip)

josesimoes, 2020-04-13:
This shouldn't be specific with text or graphs, as the driver is unaware of the buffer data contents. 
The root cause is most likely the variable type or the contents that are being sent.

Possible duplicate of #552.

Dweaver309, 2020-04-13:
I think this is a duplication of #552. If I am understanding the conversation on Discord correctly the issue may be solved? If this code is changed---
https://github.com/nanoframework/metadata-processor/blob/develop/source/native/nf-interpreter/src/CLR/CorLib/corlib_native_System_Runtime_CompilerServices_RuntimeHelpers.cpp

If this line CLR_PMETADATA ptrSrc2  = inst.m_assm->GetSignature( inst.m_target->sig );

Is added just below these lines

        CLR_PMETADATA ptrSrc  = inst.m_assm->GetSignature( inst.m_target->defaultValue );
        CLR_UINT32    lenSrc; NANOCLR_READ_UNALIGNED_UINT16( lenSrc, ptrSrc );

The array issue is solved?

josesimoes, 2020-04-13:
@Dweaver309 under investigation. Please hold.

closed 2020-04-13 by josesimoes