nanoframework/Home #596: Use of SPI APIs stops device deploy/debug after first execution

KiwiBryn, 2020-04-12:
## Details about Problem

If I call SPI API(s) in my code the application will Run/Debug once successfully, if I try to Run/Debug again the download will fail.

**Target:**
ST_STM32F429I_DISCOVERY

**Firmware image version:**
ST_STM32F429I_DISCOVERY-1.4.0-preview.155

Worked before? If so, with which nanoFramework image version:
I think it used to work trying to figure out which permutation of binaries, VS Extension & NuGets worked.

## Description
In my case as soon as I include acall to 
  using (SpiDevice device = SpiDevice.FromId("SPI5", settings))
in my code the application will Run/Debug once successfully, if I try to deploy/Debug again the download fails. 
I can download and run a simple blinky application many times without any problems. 

The only way to get device to work again is to use ST-Link too to do full erase and re-flash device.

1. ST-Link full erase

2. Use ST-Link or nanoff --target ST_STM32F429I_DISCOVERY --update to flash device.

3. Run only flashing LED with SPI code commented out (can stop & start multiple times)

Attaching to nanoDevice...
Waiting for nanoDevice to initialize...
Updating nanoDevice debugger engine.
The nanoDevice runtime is loading the application assemblies and starting execution.
'nanoFramework.Tools.VS2019.Extension.dll' (Managed): Loaded 'C:\Users\BrynLewis\source\repos\RFM9X.NetNF\ShieldSPI\bin\Debug\ShieldSPI.exe', Symbols loaded.
'nanoFramework.Tools.VS2019.Extension.dll' (Managed): Loaded 'C:\Users\BrynLewis\source\repos\RFM9X.NetNF\packages\nanoFramework.Windows.Devices.Spi.1.3.0-preview.12\lib\Windows.Devices.Spi.dll', Symbols loaded.
'nanoFramework.Tools.VS2019.Extension.dll' (Managed): Loaded 'C:\Users\BrynLewis\source\repos\RFM9X.NetNF\packages\nanoFramework.Runtime.Events.1.4.2-preview.1\lib\nanoFramework.Runtime.Events.dll', Symbols loaded.
'nanoFramework.Tools.VS2019.Extension.dll' (Managed): Loaded 'C:\Users\BrynLewis\source\repos\RFM9X.NetNF\packages\nanoFramework.Windows.Devices.Gpio.1.4.1-preview.13\lib\Windows.Devices.Gpio.dll', Symbols loaded.
The thread '<No Name>' (0x2) has exited with code 0 (0x0).
Register 0x42 - Value 0X00
Register 0x42 - Value 0X00
Register 0x42 - Value 0X00

4. Run with SPI code un-commented works as expected  
Attaching to nanoDevice...
Waiting for nanoDevice to initialize...
Updating nanoDevice debugger engine.
The nanoDevice runtime is loading the application assemblies and starting execution.
'nanoFramework.Tools.VS2019.Extension.dll' (Managed): Loaded 'C:\Users\BrynLewis\source\repos\RFM9X.NetNF\ShieldSPI\bin\Debug\ShieldSPI.exe', Symbols loaded.
'nanoFramework.Tools.VS2019.Extension.dll' (Managed): Loaded 'C:\Users\BrynLewis\source\repos\RFM9X.NetNF\packages\nanoFramework.Windows.Devices.Spi.1.3.0-preview.12\lib\Windows.Devices.Spi.dll', Symbols loaded.
'nanoFramework.Tools.VS2019.Extension.dll' (Managed): Loaded 'C:\Users\BrynLewis\source\repos\RFM9X.NetNF\packages\nanoFramework.Runtime.Events.1.4.2-preview.1\lib\nanoFramework.Runtime.Events.dll', Symbols loaded.
'nanoFramework.Tools.VS2019.Extension.dll' (Managed): Loaded 'C:\Users\BrynLewis\source\repos\RFM9X.NetNF\packages\nanoFramework.Windows.Devices.Gpio.1.4.1-preview.13\lib\Windows.Devices.Gpio.dll', Symbols loaded.
The thread '<No Name>' (0x2) has exited with code 0 (0x0).
Register 0x42 - Value 0X12
Register 0x42 - Value 0X12
Register 0x42 - Value 0X12

5. Run again with SPI code un-commented download fails
2020-04-12 12:01:02Z [Check and start debug engine on nanoDevice.]
2020-04-12 12:01:02Z [Connect successful.]
2020-04-12 12:01:04Z [Erase deployment area successful.]
2020-04-12 12:01:04Z [Device status verified as being in initialized state. Requesting to resume execution.]
2020-04-12 12:01:04Z [Waiting for device to report initialization completed (1/5).]
2020-04-12 12:01:04Z [Device reported running nanoCLR. Requesting to reboot nanoCLR.]
2020-04-12 12:01:06Z [Device has completed initialization.]
2020-04-12 12:01:06Z [Computing deployment blob.]
2020-04-12 12:01:06Z [Deploying assemblies.]
2020-04-12 12:01:06Z [Incrementally deploying assemblies to the device]
2020-04-12 12:01:07Z [Deployed 43316/43316 bytes.]
2020-04-12 12:01:07Z [Deployed assemblies with a total size of 43316 bytes.]
2020-04-12 12:01:07Z [Assemblies successfully deployed to the device.]
2020-04-12 12:01:21Z [Check and start debug engine on nanoDevice.]
2020-04-12 12:01:21Z [Connect successful.]
[... 63 lines cut ...]
2020-04-12 12:03:04Z [NanoDevices: candidate nano device \\?\USB#VID_0483&PID_5740#NANO_38673373303#]
2020-04-12 12:03:04Z [NanoDevices: checking device  \\?\USB#VID_0483&PID_5740#NANO_38673373303# @ 921600]
2020-04-12 12:03:04Z [NanoDevices: valid device ST_STM32F429I_DISCOVERY @ COM11 \\?\USB#VID_0483&PID_5740#NANO_38673373303#]
2020-04-12 12:03:04Z [NanoDevices: Serial device enumeration completed. Found 1 devices]
...

6. ST-Link full erase

7. Use ST-Link or nanoff --target ST_STM32F429I_DISCOVERY --update to flash device.

8. Blinky works again multiple times.

## Other suggested things
Maybe other SPI APIs I noticed issue with SpiDevice.FromId, The nanoFramwork sample SPI application (which is for Disco board) exhibits same behaviour.

## Expected behaviour
Should be able to download and run program many times in a session.

## Additional context
Working on LoRa library which communicates with Semtech SX127X chip via SPI.

Attached sample code and screen grab of NuGets & extension versions
[image]
[image]
[Program.zip](https://github.com/nanoframework/Home/files/4465695/Program.zip)

networkfusion, 2020-04-12:
Possibly duplicate of #563

KiwiBryn, 2020-04-14:
Possibly related to #424 and/or #447 which are also for STM boards. Refined my searching

josesimoes, 2020-04-16:
Can you confirm that this occurs only if you have VS running?

KiwiBryn, 2020-04-16:
I modified the SPI Sample code to flash onboard LED every time Gyro was successfully read.

On my Disco429 With the micro USB plugged in and VS running, the SPI sample application runs once in debugger (LED flashing approx 1 sec) , then dies at SpiDevice.FromId if I stop the application then try to debug it again.

If I then unplug the Micro USB cable (device powered by mini USB)  I think the SPI sample application starts and runs. as the LED starts to flash again. 

If I then plug the Micro USB back in, the LED stops flashing, so I assume the  SPI sample application has crashed
[Program.zip](https://github.com/nanoframework/Home/files/4486591/Program.zip)
.

josesimoes, 2020-04-16:
Thanks for confirming what I'm seeing here.

This is happening because on the device detection, VS sends a PING to the device, which sets the debugger present flags. Those make the system "think" that there is a debugger attached and therefore has to wait for the command to start execution. Because that "never" happens a timeout is triggered and the system reboots.
The end result is that with the reboot cycle happening it's difficult to VS to grab the device and running the deploy -> start debug session workflow.

closed 2020-04-21 by josesimoes

KiwiBryn, 2020-04-23:
Hi

I'm still having the same problem with the sample SPI application terminating at 

 public L3GD20(string spiBus, GpioPin chipSelectLine)
{
_chipSelectLine = chipSelectLine;
_chipSelectLine.SetDriveMode(GpioPinDriveMode.Output);

var connectionSettings = new SpiConnectionSettings(chipSelectLine.PinNumber);
connectionSettings.DataBitLength = 8;
connectionSettings.ClockFrequency = 10000000;

// create SPI device for gyroscope
**_gyroscope = SpiDevice.FromId(spiBus, connectionSettings);**
}

on the second time I debug the SPI sample on my 429Disco.

Erasing the device with STM32 ST-Link seems to fix issue.
[image]

I have updated the VS extension
[image]

Also the NuGets
[image]

Is there anything else I need todo?

@KiwiBryn

networkfusion, 2020-04-23:
Please see #563

josesimoes, 2020-04-23:
@KiwiBryn clarification: does this occur only with single step debug or always hangs?

Second: know that you can erase the deployment directly from VS with the "erase" button in device explorer.

KiwiBryn, 2020-04-23:
I'm not single stepping I have just been pressing F5. It only happens the second time I press F5. In #563 which is on ESP32 they are single stepping and it's not explicitly stated if it's every time or second time .

Yes, the button which looks like an eraser (just like ST-Link toolbar icon).  When I do that in the output window I get
2020-04-24 08:15:32Z [NanoDevices: checking device  \\?\USB#VID_0483&PID_5740#NANO_38673373303# @ 921600]
2020-04-24 08:15:32Z [NanoDevices: valid device ST_STM32F429I_DISCOVERY @ COM11 \\?\USB#VID_0483&PID_5740#NANO_38673373303#]
2020-04-24 08:15:32Z [NanoDevices: Serial device enumeration completed. Found 1 devices]
2020-04-24 08:15:54Z [Error stopping debug. Shutdown flag is set.]
2020-04-24 08:15:54Z [Error stopping debug. Shutdown flag is set.]
ST_STM32F429I_DISCOVERY @ COM11 deployment area erased.

After I erase the device from device explorer then press F5 I get 

Waiting for nanoDevice to initialize...
Updating nanoDevice debugger engine.
The nanoDevice runtime is loading the application assemblies and starting execution.
'nanoFramework.Tools.VS2019.Extension.dll' (Managed): Loaded 'C:\Users\BrynLewis\source\SPI\bin\Debug\Spi.DemoApp.exe', Symbols loaded.
'nanoFramework.Tools.VS2019.Extension.dll' (Managed): Loaded 'C:\Users\BrynLewis\source\SPI\packages\nanoFramework.Windows.Devices.Gpio.1.4.1-preview.15\lib\Windows.Devices.Gpio.dll', Symbols loaded.
'nanoFramework.Tools.VS2019.Extension.dll' (Managed): Loaded 'C:\Users\BrynLewis\source\SPI\packages\nanoFramework.Runtime.Events.1.4.2-preview.4\lib\nanoFramework.Runtime.Events.dll', Symbols loaded.
'nanoFramework.Tools.VS2019.Extension.dll' (Managed): Loaded 'C:\Users\BrynLewis\source\SPI\packages\nanoFramework.Windows.Devices.Spi.1.3.0-preview.14\lib\Windows.Devices.Spi.dll', Symbols loaded.
The thread '<No Name>' (0x2) has exited with code 0 (0x0).

Which then continues looping like in the initial issue

After some experimentation if I power cycle the 429Disco after using the erase in device explorer. I can F5 again (but sill only once).  

Thanks for your patience @josesimoes

josesimoes, 2020-04-24:
You're most welcome @KiwiBryn 😄 

I think the root cause for this has been identified. A fix it's on the way.

BTW, another alternative method to erase deployment is to keep the user button (Blue one) pressed in the STM targets during boot (reset button). This will make the boot to remain on nanoBooter without running CLR (therefore the app) thus making it accessible by VS even in the event of a faulty deployment or misbehaving app.