nanoframework/Home #599: CustomAttributes on classes get lost during metadata processing

SandorDobos, 2020-04-16:
## Details about Problem

**nanoFramework area:** Visual Studio extension (metadata-processor)

**VS extension version:** 2019.1.8.11

## Description

CustomAttributes, like TestClass get lost during "minimalize" phase of metadata processing.

## Detailed repro steps so we can see the same problem

1. Create class having one CustomAttribute on it.
2. Add some code which calls GetCustomAttributes on that class
3. Deploy to your MCU
4. During runtime you will find 0 attributes as a result of the call at step 2.

## Expected behaviour

CustomAttributes must have returned.

josesimoes, 2020-04-16:
If you run the Reflection sample, it won't work at all? Is that correct?

SandorDobos, 2020-04-16:
The Reflection sample worked :(

After stripping down all things from Reflection sample and the code where I was detecting the problem, I found the difference causing the problem:
The Reflection sample uses Shared project for Attributes, etc. Technically all classes in Shared project will be compiled into the referring project.
My code was using NF class library project which gets compiled into separate assembly.

I attached [the solution package](https://github.com/nanoframework/Home/files/4489920/Reflection.zip) to demonstrate the situation. 
It contains a simple class with one method and one field. The class is marked with TestAttribute.
The program simply prints out the customattribute count found on the class, on the method and on the field, separated with a colon.
All three projects inside have the same TestAttrs.cs file. Switching between which file's contents become uncommented you can see what happens:
- Uncommenting TestAttrs.cs in CustomAttributes or CustomAttributes.Shared projects results in "1:1:1" on console.
- Uncommenting TestAttrs.cs in NFClassLibrary1 results in "0:0:0".

But if it cant work with separate assemblies it means we cannot distribute nugets with Attribute classes defined inside because they wont be enumerated!

I wasnt sure at all that the PR I sent may solve this problem after these discoveries.
But I tried something more: I added the same attribute what I added to the MyTestClass to
a member method and the result become "1:1:0"! Same occures if you add it to MyTestClass's field,
result gets "1:0:1".

So if the attribute gets referenced on a method it gets discovered on the class too regardless it is placed in separate assembly! 
That may happen because field and method attributes are registered in metadata-processor via their constructor's metadata and the class's via their attributetype's metadata. 
So now I think again, that my PR will solve this :)

josesimoes, 2020-04-17:
Got you! 
And that makes sense. The sample Reflection is dealing with attributes _within_ the same assembly. It doesn't test for attributes on another assembly.

closed 2020-04-20 by josesimoes