nanoframework/Home #600: ST_NUCLEO64_F091RC debugger won't start

KiwiBryn, 2020-04-17:
## Details about Problem

**Target:**
ST_NUCLEO64_F091RC

**Firmware image version:**
ST_NUCLEO64_F091RC-1.4.0-preview.158

## Description
Can't run basic blinky application in debugger.

## Detailed repro steps so we can see the same problem

1. Erase device with STM32 ST Link Utility

2. Flash device with nanoff --target ST_NUCLEO64_F091RC --update (does seem to take a long time to "Reset MCU on device...")

3. Un-comment GPIO config for ST_NUCLEO64_F091RC device
  // STM32F091RC: PA5 is LED_GREEN
  GpioPin led = GpioController.GetDefault().OpenPin(PinNumber('A', 5));

5. In Visual Studio "Start Debugging"

1>------ Deploy started: Project: Blinky, Configuration: Debug Any CPU ------
1>Getting things ready to deploy assemblies to nanoFramework device: ST_NUCLEO64_F091RC @ COM14.
1>nanoDevice is initialized
1>Adding Blinky v1.0.0.0 (644 bytes) to deployment bundle
1>Adding Windows.Devices.Gpio v1.4.1.0 (5472 bytes) to deployment bundle
1>Adding nanoFramework.Runtime.Events v1.4.2.0 (3464 bytes) to deployment bundle
1>Adding mscorlib v1.7.2.0 (29236 bytes) to deployment bundle
1>Deploying 4 assemblies to device... Total size in bytes is 38816.
1>Deployment successful.

2020-04-17 18:40:54Z [Check and start debug engine on nanoDevice.]
2020-04-17 18:40:54Z [Connect successful.]
2020-04-17 18:40:54Z [Erase deployment area successful.]
2020-04-17 18:40:59Z [Computing deployment blob.]
2020-04-17 18:41:00Z [Deploying assemblies.]
2020-04-17 18:41:06Z [Deploying assemblies. Second attempt.]
2020-04-17 18:41:07Z [Incrementally deploying assemblies to the device]
2020-04-17 18:41:07Z [Deployed 2048/38816 bytes.]
2020-04-17 18:41:07Z [Deployed 4096/38816 bytes.]
2020-04-17 18:41:07Z [Deployed 6144/38816 bytes.]
2020-04-17 18:41:07Z [Deployed 8192/38816 bytes.]
2020-04-17 18:41:07Z [Deployed 10240/38816 bytes.]
2020-04-17 18:41:08Z [Deployed 12288/38816 bytes.]
2020-04-17 18:41:08Z [Deployed 14336/38816 bytes.]
2020-04-17 18:41:08Z [Deployed 16384/38816 bytes.]
2020-04-17 18:41:08Z [Deployed 18432/38816 bytes.]
2020-04-17 18:41:08Z [Deployed 20480/38816 bytes.]
2020-04-17 18:41:08Z [Deployed 22528/38816 bytes.]
2020-04-17 18:41:08Z [Deployed 24576/38816 bytes.]
2020-04-17 18:41:09Z [Deployed 26624/38816 bytes.]
2020-04-17 18:41:09Z [Deployed 28672/38816 bytes.]
2020-04-17 18:41:09Z [Deployed 30720/38816 bytes.]
2020-04-17 18:41:09Z [Deployed 32768/38816 bytes.]
2020-04-17 18:41:09Z [Deployed 34816/38816 bytes.]
2020-04-17 18:41:09Z [Deployed 36864/38816 bytes.]
2020-04-17 18:41:09Z [Deployed 38816/38816 bytes.]
2020-04-17 18:41:09Z [Deployed assemblies with a total size of 38816 bytes.]
2020-04-17 18:41:09Z [Assemblies successfully deployed to the device.]
2020-04-17 18:41:10Z [Assigning debug engine of target device.]
2020-04-17 18:41:10Z [Attempting to update device debugger flags (1/5).]
2020-04-17 18:41:10Z [Device is running CLR, requesting reboot and pause for debugger (1/5).]
2020-04-17 18:41:11Z [Device is running CLR in initialized state.]
2020-04-17 18:41:11Z [Updating information about assemblies on the debugging target.]
2020-04-17 18:41:16Z [Error loading assemblies. Assemblies count is 0.]
2020-04-17 18:41:16Z [Exception starting CLR on device: Device is reporting no assemblies deployed. Can't debug!.]

Attaching to nanoDevice...
Waiting for nanoDevice to initialize...
Updating nanoDevice debugger engine.
The nanoDevice runtime is loading the application assemblies and starting execution.
The program '[1] nanoFramework application: Managed' has exited with code 0 (0x0).
The nanoDevice target and the debugger engine failed to initialize because of unspecified device errors.
The debugger engine thread has terminated unexpectedly with error 'Device is reporting no assemblies deployed. Can't debug!'.

I can deploy to the device multiple times. I think the code is on the device as I can manually reset or power cycle the device and the onboard LED starts to flash as expected.

## Other suggested things
The device is brand new the USB driver installed works for download etc. but also maps a drive.
[Program.zip](https://github.com/nanoframework/Home/files/4491357/Program.zip)

[image]

## Expected behaviour
Should be able to download and run program many times in a session.

## Additional context
I'm using the sample blinky application as a starting point

[image]

[image]

KiwiBryn, 2020-04-17:
I unintentionally did a "Start without debugging" and this modal dialog box popped up which seemed a bit odd. 

[image]

josesimoes, 2020-04-23:
> If I do a "Start without debugging" this modal dialog box pops up which seems a bit odd.

That's a message box with dummy message that was left there during the early stages of the development. You can ignore that. The Start without debugging doesn't work anyway.

closed 2020-04-29 by josesimoes