nanoframework/Home #602: GetCustomAttributes not working

MateuszKlatecki, 2020-04-22:
## Details about Problem

**nanoFramework area:** Visual Studio extension (MDP)

**VS version:**

**VS extension version:**

**Target:**

**Firmware image version:**

**Device capabilities output:**

## Description

There are two problems:

- Custom Attributes with constructors not work for classes and methods.
- Custom Attributes with constructors not work for fields if Attribute comes from external library

## Detailed repro steps so we can see the same problem

See attached projects
[AttProblem.zip](https://github.com/nanoframework/Home/files/4515716/AttProblem.zip)

## Other suggested things

## Expected behaviour

A clear and concise description of what you expected to happen.

## Screenshot

## Additional context

Add any other context about the problem here.

josesimoes, 2020-04-23:
@MateuszKlatecki the fix with attributes constructors was implemented for fields only. Nothing with classes and methods... that's why it's failing.

MateuszKlatecki, 2020-04-24:
So if I understand correctly that there is no support for attribute constructors in System.Reflection.[*]Info (except FieldInfo) and in native methods? Is MDP ready for this?

josesimoes, 2020-04-24:
That is correct. The major changes with attribute constructors have made those available for fields (which was the task at hand). Nothing was changed in classes and methods.
And yes, the new MDP can handle all this and more.

closed 2020-04-27 by josesimoes