nanoframework/Home #613: AWS MQTT demo fails with MqttConnectionException on ESP32

awsfanatic, 2020-05-22:
### Details about Problem

**nanoFramework area:** C# code, Nuget packages

**Target:** ESP-WROOM-32

### Description

I'm trying to connect my ESP32 to AWS IoT Core. I created a thing, a certificate, keypair and a policy in AWS Console. Now I'm following AWS MQTT example from https://github.com/nanoframework/Samples/tree/master/samples/MQTT/AwsMQTT.

Before connecting to the MQTT broker I connect to the network and set current time from SNTP. So the boilerplate looks like this:

```csharp
// Connect to the Internet...
// Set RTC time and date...

// Put my own data here.
string awsHost = "a5dcq4i6ievv5-ats.iot.us-east-1.amazonaws.com"; 
string clientId = Guid.NewGuid().ToString();
string clientRsaSha256Crt =@"-----BEGIN CERTIFICATE-----...-----END CERTIFICATE-----";
string clientRsaKey =@"-----BEGIN RSA PRIVATE KEY-----...-----END RSA PRIVATE KEY-----";

// Initialize client.
X509Certificate caCert = new X509Certificate(Resources.GetBytes(Resources.BinaryResources.AwsCAroot));
X509Certificate2 clientCert = new X509Certificate2(clientRsaSha256Crt, clientRsaKey, "");
MqttClient client = new MqttClient(awsHost, 8883, true, caCert, clientCert, MqttSslProtocols.TLSv1_2);

// And finally connect.
client.Connect(clientId);
```
This is what I get while debugging:

```
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.Net.Security.SslNative::SecureConnect [IP: 0000] ++++
    ++++ System.Net.Security.SslStream::Authenticate [IP: 004c] ++++
    ++++ System.Net.Security.SslStream::AuthenticateAsClient [IP: 000a] ++++
    ++++ uPLibrary.Networking.M2Mqtt.MqttNetworkChannel::Connect [IP: 0047] ++++
    ++++ uPLibrary.Networking.M2Mqtt.MqttClient::Connect [IP: 000f] ++++
    ++++ AwsMQTT.Program::SetupMqtt [IP: 0044] ++++
    ++++ AwsMQTT.Program::Main [IP: 0025] ++++
    ++++ Exception uPLibrary.Networking.M2Mqtt.Exceptions.MqttConnectionException - 0x00000000 (1) ++++
    ++++ Message: Socket exception connecting to the broker: -16
    ++++ uPLibrary.Networking.M2Mqtt.MqttClient::Connect [IP: 003c] ++++
    ++++ uPLibrary.Networking.M2Mqtt.MqttClient::Connect [IP: 000f] ++++
    ++++ AwsMQTT.Program::SetupMqtt [IP: 0044] ++++
    ++++ AwsMQTT.Program::Main [IP: 0025] ++++
Exception thrown: 'uPLibrary.Networking.M2Mqtt.Exceptions.MqttConnectionException' in nanoFramework.M2Mqtt.dll
An unhandled exception of type 'uPLibrary.Networking.M2Mqtt.Exceptions.MqttConnectionException' occurred in nanoFramework.M2Mqtt.dll
Additional information: Socket exception connecting to the broker: -16
```

Sure looks like some SSL problem, or certificate mismatch. Changing `MqttSslProtocols.TLSv1_2` to `MqttSslProtocols.TLSv1_1` or `MqttSslProtocols.TLSv1_0` will change the socket exception from `-16` to `-29184`, but that's it. 

I double checked my certificates and keys that I downloaded from AWS Console. I also tried different library versions with no success and the same result.

### Detailed repro steps so we can see the same problem

1. Unpack the ZIP
[AwsMQTT.zip](https://github.com/nanoframework/Home/files/4670469/AwsMQTT.zip)

2. Put your WiFi SSID and password to `NetworkHelpers.cs`.

3. Build and run the project on any ESP-WROOM-32 board.

networkfusion, 2020-05-23:
AWS only supports TLS 1.2 so the others would definitely fail. You can check here for code that is known to recently connect to AWS: https://github.com/networkfusion/OrgPal-Three-Demo/blob/master/OrgPalThreeDemo/Program.cs

I am unable to check the ESP and I dont think the other devs have an aws account, so your best bet is do download the mqtt lib add it to your project and step though the offending code

networkfusion, 2020-05-25:
Confirmed working on STM32F769I with latest firmware: 
[image]

awsfanatic, 2020-05-25:
@networkfusion Thanks for the tip. I tried OrgPalThreeDemo (had to strip down all the code for the peripherals) with the same results.

I tracked it down to a call to `nanoFramework.System.Net.SslStream.AuthenticateAsClients` which in turn calls `Authenticate`: https://github.com/nanoframework/lib-nanoFramework.System.Net/blob/develop/source/nanoFramework.System.Net/Security/SslStream.cs#L131

Now I'm not sure where to go from there. What is `SslNative`?

I still think I might have done something wrong with the certificates. But I checked in the debugger that they are correctly deserialized, meaning that `caCert` has a certificate with subject `C=US, O=Amazon, CN=Amazon Root CA 1` and `clientCert` has `HasPrivateKey=1` and some certificate with `CN=AWS IoT Certificate`. That looks legit. I just copied and pasted from the files provided by AWS. Is there any postprocessing required?

networkfusion, 2020-05-25:
It might be worth checking the root cert is valid for your domain... There are a few... e.g. https://www.amazontrust.com/repository/

although I think the first one should be valid for all targets.

SslNative is the interaction with the nF firmware. It is likely not passing the cert properly?! are you using the ones generated by AWS, or your own ones (note that ECC has never been tried)...

networkfusion, 2020-05-25:
Also, can you add a Debug.WriteLine statement to ensure you have a correct DateTime and a valid IP before calling MQTT connection. given that this is more likely a question rather than an issue, I would suggest you swap to our discord group so others can help....

josesimoes, 2020-05-25:
Care to try the SSL demo with those certs?
This will help to validate if the certs are correct and/or being parsed correctly.

awsfanatic, 2020-05-25:
> It might be worth checking the root cert is valid for your domain... There are a few... e.g. https://www.amazontrust.com/repository/

I downloaded the root CA that Amazon suggested when I created the IoT "thing". Also, it matches yours root CA byte-to-byte, I compared MD5 sums.

> are you using the ones generated by AWS, or your own ones (note that ECC has never been tried)...

The ones generated by AWS. I'm not using my own CA or self-signed certificate. AWS generated those for me, I just downloaded them.

> note that ECC has never been tried

AWS doesn't provide an option to select key type in AWS Console. But it generated the "usual" RSA key:

```
# openssl rsa -text -in  6db58c147a-private.pem.key -noout
RSA Private-Key: (2048 bit, 2 primes)
...

```

> Also, can you add a Debug.WriteLine statement to ensure you have a correct DateTime and a valid IP before calling MQTT connection.

Sure, I checked before creating an issue on GitHub:
```
Network connection is: Wi-Fi
Waiting for network up and IP address...
Checking for IP
We have and IP: 192.168.31.122
Setting up system clock...
The thread '<No Name>' (0x3) has exited with code 0 (0x0).
Waiting for valid Date & Time...
Date/time: 05/25/2020 18:31:20
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.Net.Security.SslNative::SecureConnect [IP: 0000] ++++
    ++++ System.Net.Security.SslStream::Authenticate [IP: 004c] ++++
    ++++ System.Net.Security.SslStream::AuthenticateAsClient [IP: 000a] ++++
    ++++ uPLibrary.Networking.M2Mqtt.MqttNetworkChannel::Connect [IP: 004c] ++++
    ++++ uPLibrary.Networking.M2Mqtt.MqttClient::Connect [IP: 0010] ++++
    ++++ OrgPalThreeDemo.Program::SetupMqtt [IP: 0053] ++++
    ++++ OrgPalThreeDemo.Program::Main [IP: 000f] ++++
Exception thrown: 'System.Net.Sockets.SocketException' in System.Net.dll
    ++++ Exception uPLibrary.Networking.M2Mqtt.Exceptions.MqttConnectionException - 0x00000000 (1) ++++
    ++++ Message: Socket exception connecting to the broker: -16
    ++++ uPLibrary.Networking.M2Mqtt.MqttClient::Connect [IP: 0041] ++++
    ++++ uPLibrary.Networking.M2Mqtt.MqttClient::Connect [IP: 0010] ++++
    ++++ OrgPalThreeDemo.Program::SetupMqtt [IP: 0053] ++++
    ++++ OrgPalThreeDemo.Program::Main [IP: 000f] ++++
Exception thrown: 'uPLibrary.Networking.M2Mqtt.Exceptions.MqttConnectionException' in nanoFramework.M2Mqtt.dll
Socket exception connecting to the broker: -16
```

>I would suggest you swap to our discord group so others can help....

Let's do that.

networkfusion, 2020-06-06:
@josesimoes has tested this on an ESP32 and shown to work. Samples have been updated to ensure they are clear.

closed 2020-06-06 by networkfusion