nanoframework/Home #615: Error in type resolution for a managed class library

Hassmann, 2020-05-28:
### Details about Problem

**nanoFramework area:** MDP
**VS version:** 2019

**VS extension version:** 2019.2.0.1

**Target:** ESP32

**Device capabilities output:**
Assemblies:
  Workbench, 1.0.0.0
  nanoFramework.System.Text, 1.0.0.0
  mscorlib, 1.7.2.0
  nanoFramework.Runtime.Events, 1.4.2.0
  nanoFramework.Hardware.Esp32.Bluetooth, 1.0.0.0

Native Assemblies:
  mscorlib v100.4.9.0, checksum 0x65E9CC45
  nanoFramework.Runtime.Native v100.0.6.2, checksum 0x210110C2
  nanoFramework.Hardware.Esp32 v100.0.7.1, checksum 0x1B75B894
  nanoFramework.Hardware.Esp32.Bluetooth v1.0.0.0, checksum 0xE3D0163B
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.7.1, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Net v100.1.3.1, checksum 0x1118F266
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  Windows.Devices.SerialCommunication v100.1.1.1, checksum 0x82260711
  Windows.Devices.Spi v100.1.4.1, checksum 0x59B5BFC3
  Windows.Devices.Wifi v100.0.6.1, checksum 0xDF2FD922
  Windows.Storage v100.0.2.0, checksum 0x5160A7B6

### Description

A class library with its native counterpart can be deployed/flashed. 
But a debug session in VS fails with this error:

Resolving.
Resolve: unknown type: nanoFramework.Hardware.Esp32.Bluetooth.Gatt.TextCharacteristic
Error: ff000000
Waiting for debug commands...

### Detailed repro steps so we can see the same problem

1. Use the nf-interpreter fork and bluetooth classlib projects from
https://github.com/Hassmann

2. Run the Workbench project in the classlib solution

networkfusion, 2020-05-28:
Please can you add the firmware version and whether you are using nanoff to deploy. If nanoff is used, and you have the capability to create local firmware, it would be good to know if using debug release causes the same issues (as minsizerelease could be optimising)

Hassmann, 2020-05-28:
I am constantly rebasing on nf-framework/development. 
Build and deploy via VS Code: Debug NanoCLR
In VS19 the error occurs in Debug and Release

josesimoes, 2020-06-05:
@Hassmann have you tried this after making the changes I've suggested in Discord and using the latest version of the VS extension?

josesimoes, 2020-08-26:
Closing as no further details have been provided to properly reproduce the issue.

closed 2020-08-26 by josesimoes