nanoframework/Home #616: httpWebRequest.GetResponse() not work after third attempt

andceoynnova, 2020-05-29:
### Details about Problem

**nanoFramework area:** (C# code)

**VS version:** (2019)

**VS extension version:** 2019.2.0.1

**Target:** ESP32

**Device capabilities output:**

System Information
HAL build info: nanoFramework running @ ESP32
  Target:   ESP32_WROOM_32
  Platform: ESP32

Firmware build Info:
  Date:     May 27 2020
  Type:     MinSizeRel build with IDF v3.3.1
  Version:  1.4.0.204
  Compiler: GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

AppDomains:

Assemblies:
  esp32_nf_Orved, 1.0.0.0
  nanoFramework.Runtime.Native, 1.2.1.0
  System.Net, 1.4.1.0
  nanoFramework.Runtime.Events, 1.4.2.0
  nanoFramework.Hardware.Esp32, 1.2.1.0
  nanoFramework.System.Text, 1.0.0.0
  Windows.Storage, 1.2.0.0
  System.Net.Http, 1.2.2.0
  Windows.Devices.SerialCommunication, 1.2.1.0
  nanoFramework.Json, 1.4.1.0
  mscorlib, 1.7.2.0
  nanoFramework.System.Collections, 1.0.1.0
  Windows.Storage.Streams, 1.8.2.0
  Windows.Devices.Wifi, 1.2.0.0

Native Assemblies:
  mscorlib v100.4.9.0, checksum 0x65E9CC45
  nanoFramework.Runtime.Native v100.0.6.2, checksum 0x210110C2
  nanoFramework.Hardware.Esp32 v100.0.7.1, checksum 0x1B75B894
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xA5C172BD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.7.1, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.4, checksum 0x39DA4F21
  System.Net v100.1.3.1, checksum 0x1118F266
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  Windows.Devices.SerialCommunication v100.1.1.1, checksum 0x82260711
  Windows.Devices.Spi v100.1.4.1, checksum 0x59B5BFC3
  Windows.Devices.Wifi v100.0.6.1, checksum 0xDF2FD922
  Windows.Storage v100.0.2.0, checksum 0x5160A7B6

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe436c  0x0001bc00
  FLASH 0x08000000  0x00800000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x180000     nanoCLR
      1    0x00190000       1      0x1B0000     Deployment
      2    0x00340000       1      0x100000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x00340000    0x100000 (1024kB)   Configuration
  0x00010000    0x180000 (1536kB)   nanoCLR
  0x00190000    0x1B0000 (1728kB)   Deployment

Deployment Map
Empty

### Description

I start from sample project HTTP and i insert code of reading of https inside a timer that it call every 1 minute.
During the get response of a https url launch a exception ever time after **third** reading.

### Expected behaviour

It must work every time the event is called.

#### Sample Project
```
Timer sendCloud_t = new Timer(SendCloudTimerCallback, null, 1000, 60000);
private void SendCloudTimerCallback(object state)
{
            X509Certificate rootCACert = new X509Certificate(letsEncryptCACertificate);
            //test URL for Let's encrypt certificate
            string url = "https://www.howsmyssl.com";
            Console.WriteLine($"Performing Http request to: {url}");

            using (var httpWebRequest = (HttpWebRequest)WebRequest.Create(url))
            {
                httpWebRequest.Method = "GET";

                //////////////////////////////////////////////////////////////////////
                // need to set the SSL protocol that the connection is going to use //
                // *** this MANDATORY otherwise the authentication will fail ***    //
                //////////////////////////////////////////////////////////////////////
                httpWebRequest.SslProtocols = System.Net.Security.SslProtocols.Tls11;

                // if the request is to a secured server we need to make sure that we either:
                // 1. provide the root CA certificate 
                // 2. the device has already stored a root CA bundle that will use when performing the authentication
                httpWebRequest.HttpsAuthentCert = rootCACert;

                using (var stream = httpWebRequest.GetResponse().GetResponseStream())
                {
                    // read response in chunks of 1k

                    byte[] buffer = new byte[1024];
                    int bytesRead = 0;

                    Console.WriteLine("Http response follows");
                    Console.WriteLine(">>>>>>>>>>>>>");

                    do
                    {
                        bytesRead = stream.Read(buffer, 0, buffer.Length);
                        Console.Write(Encoding.UTF8.GetString(buffer, 0, bytesRead));
                    }
                    while (bytesRead >= buffer.Length);
                }

                Console.WriteLine(">>>>>>>>>>>>>");
                Console.WriteLine("End of Http response");
            }
}
```

### Screenshot
[image]

networkfusion, 2020-05-29:
2 things, why are you not using TLS1.2 and, what happens if you wrap in a try, catch loop? do all future calls fail to succeed?

andceoynnova, 2020-06-03:
I try TLS 1.1 and 1.2 but nothing changes...
I just try wrap in a try catch in any case after the third launch the exception.

josesimoes, 2020-06-04:
@andceoynnova communication failures can occur....

Does it work after the 3 attempt, if wrapped on a try catch? Or it just fails "forever" after the 3rd attempt?

andceoynnova, 2020-06-05:
I update to last version of firmware and it fails "forever" after the 2nd attempt.
```
Performing Http request to: https://www.howsmyssl.com
    ++++ Exception System.Net.Sockets.SocketException - CLR_E_FAIL (4) ++++
    ++++ Message: 
    ++++ System.Net.Security.SslNative::SecureClientInit [IP: 0000] ++++
    ++++ System.Net.Security.SslStream::Authenticate [IP: 003d] ++++
    ++++ System.Net.Security.SslStream::AuthenticateAsClient [IP: 000a] ++++
    ++++ System.Net.HttpWebRequest::EstablishConnection [IP: 0238] ++++
    ++++ System.Net.HttpWebRequest::SubmitRequest [IP: 0019] ++++
    ++++ System.Net.HttpWebRequest::GetResponse [IP: 000c] ++++
    ++++ HttpSamples.HttpWebRequestSample.Program::SendCloudTimerCallback [IP: 003a] ++++
Eccezione generata: 'System.Net.Sockets.SocketException' in System.Net.dll
Eccezione generata: 'System.Net.Sockets.SocketException' in System.Net.dll
    ++++ Exception System.Net.WebException - 0x00000000 (4) ++++
    ++++ Message: 
    ++++ System.Net.HttpWebRequest::GetResponse [IP: 00c8] ++++
    ++++ HttpSamples.HttpWebRequestSample.Program::SendCloudTimerCallback [IP: 003a] ++++
Eccezione generata: 'System.Net.WebException' in System.Net.Http.dll
Error...
>>>>>>>>>>>>>
End of Http response
```

josesimoes, 2020-06-05:
I've reproduced this.

It starts failing on the call mbedtls_ssl_setup to with a -0x7F00error (Memory allocation failed).
From there on it goes on failling early and early on the previous calls to mdebtls API. Eventually the system crashes.
Most likely there is a memory leak which ends up on heap starvation.

jefmad, 2020-07-15:
Hello all,

I’ve been facing the same issue with the last preview and I remember it was not the case when developing my project in March. **When downgrading to this build ESP32_WROOM_32-1.4.0-preview.107, the issue is not present**, and the https requests are correctly performed.
I have an ESP32 hosting a web server and at the same time connecting to internet to https sites every 30 sec. It has been working for one week with the following conf.
Once, I noticed that after adding nanoFramework.Hardware.Esp32 to my project I got the issue described above, but I’ve not been able to reproduce the error. I’m just adding this last information if that could help...

**VS Extension** : nanoFramework.Tools.VS2019.Extension 1.4.0.2

System Information
HAL build info: 1.4.0.0, nanoFramework running ESP32 built with IDF v3.3.1
Target: ESP32_WROOM_32
Platform: ESP32

Image build @ Feb 17 2020 GNU ARM GCC v5.2.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Solution Build Info: 1.4.0.0, nanoFramework running ESP32 built with IDF v3.3.1

AppDomains:

Assemblies:
  NFApp1, 1.0.0.0
  System.Net, 1.4.1.0
  Windows.Devices.Gpio, 1.4.1.0
  Windows.Devices.Pwm, 1.5.1.0
  nanoFramework.System.Text, 1.0.0.0
  Windows.Storage.Streams, 1.8.0.0
  nanoFramework.Runtime.Native, 1.2.0.0
  mscorlib, 1.6.0.0
  Windows.Devices.Adc, 1.4.1.0
  nanoFramework.Hardware.Esp32, 1.2.0.0
  System.Math, 1.2.0.0
  nanoFramework.Devices.OneWire, 1.2.0.0
  nanoFramework.Networking.Sntp, 1.4.0.0
  nanoFramework.System.Collections, 1.0.0.0
  System.Net.Http, 1.2.2.0
  Windows.Storage, 1.2.0.0
  nanoFramework.Runtime.Events, 1.4.1.0

Native Assemblies:
  mscorlib v100.4.6.0, checksum 0x3E5F09B0
  nanoFramework.Runtime.Native v100.0.6.0, checksum 0x035EC7E8
  nanoFramework.Hardware.Esp32 v100.0.7.0, checksum 0xC16385A2
  nanoFramework.Devices.OneWire v100.0.3.3, checksum 0xC7BEAEBC
  nanoFramework.Networking.Sntp v100.0.4.3, checksum 0xB617FDFB
  nanoFramework.ResourceManager v100.0.0.0, checksum 0xCDF9E76C
  nanoFramework.System.Collections v100.0.0.0, checksum 0xC7167094
  nanoFramework.System.Text v100.0.0.0, checksum 0x482137FC
  nanoFramework.Runtime.Events v100.0.7.0, checksum 0x7B4AC2F2
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.4.3, checksum 0xC5A2A195
  System.Net v100.1.3.0, checksum 0x0B718489
  Windows.Devices.Adc v100.1.3.2, checksum 0xDEDF1136
  Windows.Devices.Gpio v100.1.2.0, checksum 0x2432E7EB
  Windows.Devices.I2c v100.2.0.1, checksum 0x8D1C8F95
  Windows.Devices.Pwm v100.1.3.2, checksum 0x84D8B10C
  Windows.Devices.SerialCommunication v100.1.1.0, checksum 0x8E46E2E2
  Windows.Devices.Spi v100.1.4.0, checksum 0xEA555AF2
  Windows.Devices.Wifi v100.0.6.0, checksum 0x5AC050C9
  Windows.Storage v100.0.1.0, checksum 0x8D8C8026

++++++++++++++++++++++++++++++++
::        Memory Map          ::
++++++++++++++++++++++++++++++++
  Type     Start       Size
[... 42 lines cut ...]

**or**
1>------ Deploy started: Project: NFApp1, Configuration: Debug Any CPU ------
1>Getting things ready to deploy assemblies to nanoFramework device: Silicon Labs CP210x USB to UART Bridge (COM5) @ COM5.
1>nanoDevice is initialized
1>Adding NFApp1 v1.0.0.0 (32752 bytes) to deployment bundle
1>Adding System.Net v1.4.1.0 (18544 bytes) to deployment bundle
1>Adding Windows.Devices.Gpio v1.4.1.0 (5476 bytes) to deployment bundle
1>Adding Windows.Devices.Pwm v1.5.1.0 (2792 bytes) to deployment bundle
1>Adding nanoFramework.System.Text v1.0.0.0 (5820 bytes) to deployment bundle
1>Adding Windows.Storage.Streams v1.8.0.0 (6528 bytes) to deployment bundle
1>Adding nanoFramework.Runtime.Native v1.2.0.0 (1588 bytes) to deployment bundle
1>Adding mscorlib v1.6.0.0 (29208 bytes) to deployment bundle
1>Adding Windows.Devices.Adc v1.4.1.0 (2772 bytes) to deployment bundle
1>Adding nanoFramework.Hardware.Esp32 v1.2.0.0 (3852 bytes) to deployment bundle
1>Adding System.Math v1.2.0.0 (1100 bytes) to deployment bundle
1>Adding nanoFramework.Devices.OneWire v1.2.0.0 (800 bytes) to deployment bundle
1>Adding nanoFramework.Networking.Sntp v1.4.0.0 (484 bytes) to deployment bundle
1>Adding nanoFramework.System.Collections v1.0.0.0 (4000 bytes) to deployment bundle
1>Adding System.Net.Http v1.2.2.0 (39044 bytes) to deployment bundle
1>Adding Windows.Storage v1.2.0.0 (5496 bytes) to deployment bundle
1>Adding nanoFramework.Runtime.Events v1.4.1.0 (3468 bytes) to deployment bundle
1>Deploying 17 assemblies to device... Total size in bytes is 163724.

Hope this could help to troubleshoot…

AdrianSoundy, 2020-07-16:
The amount of ram allocated to managed heap on ESP32 was increased at some point. Maybe this is causing the issue. May also just be a memory leak issue.

josesimoes, 2020-08-25:
Barking at the wrong tree... Turns out that the root cause was in the HTTP library, specifically in the HttpWebRequest and Response. HttpWebResponse was missing  the `Close()` method. HttpWebRequest was initializing the input network stream wrongly for HTTPS connections.

Plus take a look at the updated HTTP WebRequest sample. There is now an improved version that's using the `using` statement properly.

closed 2020-08-25 by josesimoes