nanoframework/Home #619: Replace SPIFFS with littleFS

networkfusion, 2020-05-30:
Mark SPIFFS as depreciated (off by default) and use littleFS instead (still an option in firmware, although not selected by default). see https://github.com/esp8266/Arduino/pull/7263 for some good reasons why.

josesimoes, 2020-06-02:
Link to the [littleFS](https://github.com/ARMmbed/littlefs)

josesimoes, 2020-06-02:
Here's the results of a direct comparison between the two file systems: https://github.com/RIOT-OS/RIOT/pull/8316

networkfusion, 2020-06-04:
Work on this can be found here: https://github.com/nanoframework/nf-interpreter/tree/develop-littlefs

networkfusion, 2020-06-05:
Note that chibiOS 20.3.x also supports EFlash subsystem (EFL) http://www.chibios.com/forum/viewtopic.php?f=3&t=4323&sid=15a7802a3245dccb6fe93526752e422a&start=40

closed 2020-08-11 by stale[bot]

josesimoes, 2022-06-21:
No work has been done to make this change. If you would like to tackle this, please go ahead. We ❤️ PRs!! 😉

networkfusion, 2022-06-22:
Here was my stab at it: https://github.com/nanoframework/nf-interpreter/commit/df3ba3fa104310317f0aa101615622c257d97eef