nanoframework/Home #620: Depreciate/drop `Windows.Devices` namespace in favour of `System.Device`

networkfusion, 2020-05-30:
This brings it inline with expected .netCore 5 namespaces and would aid compatibility towards .Net Standard support

alberk8, 2020-06-09:
I second the change.

networkfusion, 2020-08-03:
PR for DAC https://github.com/nanoframework/lib-System.Devices.Dac/pull/24

networkfusion, 2020-08-03:
PR for Gpio https://github.com/nanoframework/lib-Windows.Devices.Gpio/pull/134

Ellerbach, 2020-09-13:
And would be great if the API can aligned with https://github.com/dotnet/iot/tree/master/src/System.Device.Gpio. That would make it easier to switch and share some code between both nanoFramework and .NET IoT.

josesimoes, 2020-09-13:
> And would be great if the API can aligned with https://github.com/dotnet/iot/tree/master/src/System.Device.Gpio. That would make it easier to switch and share some code between both nanoFramework and .NET IoT.

That's exactly the plan. Now that we've decided to move away from the UWP API, that's the next  one we plan to adhere to.

Are you willing to give us a hand with this?

This is not a challenging task, just that it takes time, has most methods just need renaming and others parameters tweaking. Not to mention the documentation comments.
Anything needed to be changed at the native end we can take care of it. 

Let me know. 🙂

krwq, 2020-09-14:
will you provide nano version of the System.Device.Gpio package (perhaps with some stuff stripped off)? I'm in favor of having IoT repo sharing some code with this if so (there are likely only few classes which need sharing) but we'd need to add some kind of CI to make sure we don't break anything by accident in the future. We're fine with adding some ifdefs in the code or something to make it easier for nano. Would be nice to be able to get Iot.Device.Bindings (or at least large part of it) to work on nano

Ellerbach, 2020-09-14:
> Are you willing to give us a hand with this?

Yes and as @krwq mentionned, we need to decide what to keep and what to remove also the interfaces. A lot of the bindings we have in .NET IoT are using Span and the functions from nanoFramework are all byte[]. Later today as soon as I'll have a little bit of time I'll publish the work I've been doing with Span and simple wrapper over I2C and Gpio. So a large part of the work is already done as well.

Ellerbach, 2020-09-14:
@josesimoes so posted the last week evening and weekend work here: https://github.com/Ellerbach/nanoFrameworkIoT

josesimoes, 2020-09-14:
@krwq @Ellerbach that's awesome! 👏 

I just did a quick check on the GPIO classes on IoT Core and that's pretty much as I remember it. Which is good as most of it is the same or very close to what we have.

Here's my suggestion: baby steps. Let's start with GPIO before moving to others (I2C, etc) this can be a PoC and we can learn from the process what's the best way to tackle it.
Any changes required at the native end: we can accommodate that easily. 

On the code sharing: that would be great. Let's find out what's the best and efficient way to make that happen. That doesn't prevent that work is started on the API migration.

josesimoes, 2020-11-21:
Very much active

josesimoes, 2022-02-09:
### ⚠️  ALL WINDOWS.DEVICES libraries will have one last stable release published on the next release cycle.
Following which the repositories will be archived and the libraries won't receive any further updates.

networkfusion, 2022-04-19:
Note that the `Can` namespace is still behind...

closed 2022-04-26 by josesimoes